PiSCSI/piscsi #514: Shutting down/Rebooting the Pi via the RaSCSI remote interface?

uweseimet, 2021-12-09:
I wonder wheter it is useful to shut down or reboot the Pi with a new remote interface operation. This may depend on the usage scenario. If one mainly uses the Pi for RaSCSi (started by systemd) and wants to shut it down, currently one has to use the command line (or some desktop functionalitiy) for this. This means you always need keybard/monitor-based interaction, even when using the RaSCSI Control app. At least when using the app it may be nice if you just start your Pi, and after switching off the computer RaSCSI is connected to you can also shut down RaSCSI without needing a monitor and keyboard

@rdmark Any thoughts on this?

uweseimet, 2021-12-09:
I thought I'd just implement this change in order to see whether it works as expected. Feature branch is feature_shutdown. There is a new "mode" argument to the SHUT_DOWN operation, "mode" can be "rascsi" (this is the previous behavior), "all" or "shutdown".

rdmark, 2021-12-09:
@uweseimet I love this idea. The webapp currently passes shutdown or reboot shell commands to the same effect, but I think your solution would be cleaner and more portable.

BTW, the "all" argument means shut down rascsi first, then the Linux system itself? I wonder if "system" would be a more easy to understand name for it.

uweseimet, 2021-12-09:
@rdmark Yes, that's the meaning of the "all" argument. It flushes the RaSCSI data and then shuts down the system. An explicit separate RaSCSI shutdown is not required. I agree, "system" would be a better name. I just pushed the required change.

In general, webapp functionality that may be useful in general might always be a candidate for the RaSCSI C++ core. Not everything can be included in the core, of course. But functionality that is useful for both the webapp and the Android app is potentially a candidate, because it is likely that other clients (rasctl, but also others, you never know) might profit from it.

akuker, 2021-12-10:
I love this idea! Thanks guys!

rdmark, 2021-12-10:
One fun fact is that this change brings feature parity with RASCTL.X for the X68k, which introduced this feature in Gimons v1.52 :)

rdmark, 2021-12-14:
@uweseimet I've been testing this through the Web Interface, and having the issue where the "system" mode doesn't actually halt the system, but leads to an effective reboot. The other two modes seems to work as intended, though.

This is on a Pi3 with Bullseye / gcc 10.

Testing through the code in this PR: https://github.com/akuker/RASCSI/pull/522

I wonder if this is a Linux kernel version or gcc library version discrepancy with your environment, if it works for you?

rdmark, 2021-12-14:
Never mind -- this was due to a silly mistake in my code. All is well!

closed 2021-12-14 by rdmark