PiSCSI/piscsi #519: Support for creating ISOs from downloads

adamathefrog, 2021-12-12:
Similar to the HFS feature, creating ISOs from downloads would be nice, perhaps FAT/FAT32 too.

Ancilliary ideas would be to unrar/zip an upload and create a filesystem from the contents, providing some flexibility being able to create a filesystem with multiple files in without having to unrar/zip on sub-zero MHz CPUs :)

rdmark, 2021-12-12:
I think this is a good idea, and genisoimage has the capabilities to do much more than the HFS extension to ISO 9660 that we use now.

Now, I don't think there is such a thing as FAT/FAT32 file systems on CD-ROM. What we have is several "levels" of ISO 9660, with various extensions for particular vendors. For instance, ISO 9660 Level 1 has an 8.3 file name format to align with 'classic' DOS limitations. Higher levels have support for longer file names and larger file sizes. What we need to figure out here is which file system formats that would be potentially useful for the RaSCSI userbase.

References:
https://en.wikipedia.org/wiki/ISO_9660
https://docs.microsoft.com/en-us/windows/win32/imapi/disc-formats

rdmark, 2021-12-12:
A first jab at implementing this: https://github.com/akuker/RASCSI/pull/520

rdmark, 2021-12-13:
> Ancilliary ideas would be to unrar/zip an upload and create a filesystem from the contents, providing some flexibility being able to create a filesystem with multiple files in without having to unrar/zip on sub-zero MHz CPUs :)

This is an interesting idea. Not sure about rar since it is a proprietary format with no good OSS solutions, but unzipping is easy enough with existing code. I'll play around with this and see how easily it can be done.

rdmark, 2021-12-14:
PR for unzipping zip files before creating iso: https://github.com/akuker/RASCSI/pull/525

rdmark, 2021-12-21:
@adamathefrog The two features discussed have been merged to develop now, which I think is enough scope for this ticket. Please raise new tickets for further improvement requests!

closed 2021-12-21 by rdmark