PiSCSI/piscsi #527: Subdirectory support for the images folder

bzeiss, 2021-12-15:
Enhancement proposal

# Info

- Which version of Pi are you using: pi3/pi4/pi zero 2
- Which github revision of software: develop
- Which board version: 2.4
- Which computer is the RaSCSI connected to: Yamaha EX5R, EMU 5000 Ultra, Roland S-760, DOS/Win machines with Adaptec SCSI controllers

# Describe the issue
It would be really awesome if RASCSI was able to support subfolders within the image subfolder. Use: the use case where RACSI really shines is it's ability to be a cdrom jukebox with it's web interface. This is what makes it attractive for samplers. Sampler users have lots of cdroms (like near maybe near 50-60 in my case) on a single micro sd card (512gb). The same use case applies, for example, for dos/windows users. Lot's of cdroms with the wish to easily swap the cdroms. 

The way it is currently organized (all in one folder) makes it hard to keep everything organized when using a lot of cdrom images. It would be an awesome great start if RASCSI and it's web interface could at least scan the images folder recursively and use all images from there in a single list. A second stage of improvement could maybe be some kind of hiearchical table of contents with links to each folder at the top to navigate the list of images. Such a ToC would likely sill work on Mac browsers.

uweseimet, 2021-12-17:
@bzeiss https://github.com/akuker/RASCSI/issues/527 might be what you are asking for, at least as far as the rascsi core is concerned.

bzeiss, 2021-12-17:
This is awesome, have to try it out! Thanks!

uweseimet, 2021-12-19:
@rdmark Correct me if I am wrong, but for the rascsi core the work is IMO done. It's up to the client to structure the files, so I would like to make this ticket a web interface ticket.
(In the app I will most likely offer a structured view in an upcoming version.)

rdmark, 2021-12-22:
@uweseimet Yes, at a glance only webapp work remains. Once the PR gets merged to develop, I'll overhaul the webapp code.

rdmark, 2021-12-23:
@bzeiss Can you please test the code in https://github.com/akuker/RASCSI/pull/558 ?
I've fixed all the bugs that you reported, except for the one where backslash combinations were treated as special characters in the Rename dialog. I tried various approaches to escape the strings, but it had unwanted side effects. I don't think backslashes in file names is really a valid usecase anyways, so I suggest we keep it as is. Seems mostly harmless. (Famous last words?)

bzeiss, 2021-12-23:
Sure, I'll give it a test drive. I agree with the backslashes. It is pretty harmless.

bzeiss, 2021-12-23:
- Creation of images at depth 1 seems to work fine with all kinds of nasty things I have thrown at it (double slashes, backslashes, spaces, existing filename, existing/non-existing directory etc.)
- Attachments of nested files are shown correctly now.
- Scan depth is shown correctly for me. 
- Renaming is still fine for me.

Changes look good to me!

rdmark, 2021-12-23:
All PRs merged. Closing this ticket.

closed 2021-12-23 by rdmark