PiSCSI/piscsi #528: Add authentication by token to rascsi core

uweseimet, 2021-12-16:
RaSCSI is not security-critical, but adding a decent level of authentication to the RaSCSI core might be useful.

Suggested approach: An additional optional parameter that denotes the full path of a file with a token string can be passed to rascsi. This file must be readable by root only and RaSCSI has to check this when it is started. All RaSCSI operations are only executed if a new "token" parameter is passed by the client with each operation. As we have named parameters, this additional parameter can be added without losing backwards compatibility.
The standard rascsi setup should IMO not require a token, i.e. additional is optional and disabled because of backwards compatibility.

Where the client stores its token is client-specific. The client does not use the same source for the token as the server, of course. All clients except the web ui may be running on a different machine anyway, don't have access to the token file, and are not running with root permissions, i.e. would not be able to read the file.
The rasctl client requires user input for the token string. The web ui might also require user input, or could be configured with a token or a token file of its own, or would not use this feature at all. Other clients might store the token (entered by the user once) in a database, e. g. the RaSCSI Control app.

This approach has a loophole, though: The password has to be sent in cleartext via the remote interface. But as already mentioned, RaSCSI is not security-critical, and any more elaborate solution is probably too complicated to be implemented at all.

rdmark, 2021-12-17:
@uweseimet To continue a discussion thread from the original ticket: The reason I suggested the Web UI to read the same token as RaSCSI proper, is that it is always expected to run on the same system as RaSCSI, and that it already has some elevated permissions to perform certain other system commands. If it was to be allowed to read the token file directly, what is the worst that can happen? Perhaps a malicious attacker getting control of the Python runtime and reading the data structures directly?

uweseimet, 2021-12-17:
@rdmark Reading the same file is the same as having no authentication on the rascsi core level for the web ui at all. All in all, the web ui is a rascsi client, and it should be treated as consistent as possible.

closed 2021-12-19 by uweseimet