PiSCSI/piscsi #533: Add interface operation that returns operation meta data

uweseimet, 2021-12-17:
A new interface operation OPERATION_META_DATA shall be added. This operation shall return a map of supported operations, their supported parameters, and whether a parameter is optional or not.
This feature will reduce the client dependency on the RaSCSI version, because a client knows about supported operations without having to rely on (hard-coded in the client) RaSCSI versions.

rdmark, 2021-12-19:
@uweseimet Do I understand this feature correctly with, say, this scenario: The very latest Android app trying to talk to a somewhat old RaSCSI server, and dynamically hides UI options for features that the server does not understand.

uweseimet, 2021-12-19:
@rdmark This is useful for any client, but the web ui is least likely to profit.
A client can query rascsi for the operations supported, and can react on it. Typical scenario: A client is aware of the SHUT_DOWN operation (and offers it to the user), but is connected to an old rascsi version that does not support this operation. In this case the client should not offer the SHUT_DOWN operation at all, because it will definitely fail. Checking the rascsi version in order to find out which operation is supported is not that good, because sooner or later the client ends up with checking many versions. And if, in a future version of rascsi, an operation is not available anymore, it gets even more difficult when using version numbers.
Instead, the client can use OPERATION_INFO to explicitly find out the operations supported by the rascsi server the client is currently connected to. (Future versions of my app will most likely support a selectablle set of rascsi instances, which is useful for users who have more than one board.) Up to a certain degree a client can even use operations not yet supported by this client, because it knows operation names and parameters. But this use is rather exotic.

uweseimet, 2021-12-20:
@rdmark Just a note: There is one more feature that does not depend on the RaSCSI version, but on how RaSCSI was started, namely the shutdown/reboot. When started with sudo all modes (rascsi, shutdown, reboot) are available. But when not started as root (unusual case, but as you know I do most of my testing on a PC, where you do not need to be root because no RaSCSI hardware is present), the meta data signal that only the "rascsi" shutdown mode is available. A client can react on this by not offering the unavailable shutdown modes.

closed 2021-12-22 by uweseimet