PiSCSI/piscsi #535: Add BIN/CUE support to RaSCSI

akuker, 2021-12-17:
# Info

- Which version of Pi are you using: 21.11
- Which github revision of software: N/A
- Which board version: N/A
- Which computer is the RaSCSI connected to: N/A

# Describe the issue

Currently in the code, BIN/CUE support is incomplete. Currently, it will throw an exception if someone tries to use it.
[image]

This issue is to request that RaSCSI be updated to support BIN/CUE disk images fully. This may also drive changes into the web interface to support them.

akuker, 2021-12-17:
Note that there are a few existing libraries for parsing CUE files. We shouldn't need to implement our own version of it. Examples:

https://github.com/coolerfall/c-cuelib (Preferred - LGPL)
https://github.com/lipnitsk/libcue (non-preferred - GPL)

TaehyunKim, 2023-05-12:
To use cdemu , it is better to support passthrough scsi generic device ( /dev/sgXX ). CDEMU provide fuctionality to emulate cdrom image to scsi generic devices. then piscsi read and forward scsi command to scsi generic devices. this method provide USB CD-ROM/DISK over PiSCSI also.