PiSCSI/piscsi #539: Add folder filter to DEFAULT_IMAGE_FILES_INFO

uweseimet, 2021-12-19:
The current filder is just for filenames, but in order to support more usage scenarios an additional filter for the folder names should be added.

rdmark, 2021-12-20:
@uweseimet I didn't ask when you introduced the filename filters, but I wanted to confirm the expected behavior: The filter here is a string of alphanumeric chars, and there are not special characters such as wildcards implemented, right? I don't think so looking at the code, but wanted to double check.

BTW, did you have particular scenarios in mind when users would want to filter files in the images dir? Off the top of my head, I could think of filtering a particular subdir, or filtering by file endings if you only want .iso files to be listed, say. But there seemed quite narrow cases.

uweseimet, 2021-12-20:
@rdmark Yes, this is just a string, not a regex. Regarding the filtering if files, if you have a lot of files in a folder you may want to filter them by filename. Or by extension, which is also covered by the filename. I thought about introducing one more filter, for the extension only, but I decided against it. (This can be added any time if need should be.)
Clients are not forced to support filtering by filename, i.e. if you don't see a need for this in the web ui, you can just not offer this option. Or you can always set both filters to the same pattern, which is effectively the same as having a single filter for the complete filename, including the folder part.

closed 2021-12-22 by uweseimet