PiSCSI/piscsi #552: Provide Localization Options for Popular Languages

akuker, 2021-12-21:
# Info

- Which version of Pi are you using: 21.11

# Describe the issue

Currently, the web interface for RaSCSI is only available in English. A useful feature to add in the future would be support different languages. A drop-down menu could potentially be added to the top of the page to select the UI language.

uweseimet, 2021-12-21:
This feature should be added (also) to the RaSCSI core (some work has already been done in the protobuf interface, in order to not have a mixture of English texts (error messges returned by rascsi) and a non-English UI.

The preferred browser language should be used, or at least should be the default setting. In other words, if there is a menu item at all, the user should not be forced to use it in order to get her/his preferred language.

There is a general maintenance problem, though: If somebody adds supportft for a new language, it is not unlikely that this person loses interest in the project or does not want to update translations any time new ui elements are added. Very soon you will end up with translations missing, and as a result the ui will look unprofessional, with a mixture of English and another language.
So think about whether you would really like to have this feature twice ;-).

rdmark, 2021-12-23:
I'm considering using the [Flask-Babel library ](https://flask-babel.tkte.ch/) to internationalize the web UI. It is basically a wrapper for gettext with a bunch of features built on top. It would do the typical gettext thing of parsing flagged strings into a messages.pot file, which you then duplicate into a messages.po file for each locale. Contributors would work with the respective po file and make PRs against develop, for instance. I'm sure there are other tooling on top of gettext that can streamline this process.

To address Uwe's maintenance concern there: I agree completely that it's very jarring to see a partially localized UI. I propose we have it as part of the release process, or perhaps dynamically on runtime, a 100% completion threshold, and only ever let completed translations be exposed to end users. Thoughts?

uweseimet, 2021-12-23:
The problem I see with this is that missing translations may block further development in the core. New features in the rascsi core may require new error messages. If a translation is missing for one of those the new feature would not be releasable if a full set of translations is mandatory.

akuker, 2021-12-23:
I would also propose that we limit internationalization to the web interface only. I think we're safe to keep the log files in English. That will also help reduce the number of translations we need.

rdmark, 2021-12-23:
@uweseimet I was proposing having the opposite policy: If a localization is not 100%, it gets disabled in the release tag. A crude way to do this if, say, the Swedish localization is behind on translating new strings, we remove 'sv' from:
`supported_languages = { "en", "de", "sv" };`

rdmark, 2021-12-23:
> I would also propose that we limit internationalization to the web interface only. I think we're safe to keep the log files in English. That will also help reduce the number of translations we need.

Yes, this is effectively what we're doing now. Uwe has applied this only on error strings that are exposed to the clients.

akuker, 2021-12-23:
My thoughts are that the logs are going to be the most useful for developers. The code is only in English, so I'm assuming the developers have some English proficiency. If I were to get a log file in French (as an example), it would be much more difficult for me to troubleshoot. Its also adding more complexity to the "core" functionality of RaSCSI.

uweseimet, 2021-12-23:
Note that the error messages are not just in the logs, but are also returned to the client.

rdmark, 2021-12-26:
Full i18n of the web interface, and a complete Swedish localization: https://github.com/akuker/RASCSI/pull/564

rdmark, 2021-12-27:
Make the language configurable in the Web UI: https://github.com/akuker/RASCSI/pull/568

rdmark, 2021-12-28:
Pretty much everything in the web app frontend is now ready for i18n, ready for others to add localizations beyond Swedish.

Spun off a new ticket for the web server error page i18n: https://github.com/akuker/RASCSI/issues/575

closed 2021-12-28 by rdmark