PiSCSI/piscsi #557: Macintosh SE stuck in boot loop when RaSCSI is the only device on the SCSI Bus

edrowley951, 2021-12-21:
# Info

- Which version of Pi are you using: ZeroW
- Which github revision of software: Release 
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: Macintosh SE

# Describe the issue
The computer will only boot into attached hard drive image if the RaSCSI is attached to the back of a Iomega Zip Drive first.  If the only SCSI device attached to the computer is the RaSCSI, the Happy Mac is shown for a few seconds and then then disappears and shows the mouse cursor.  This repeats indefinitely.  Happens with any of the HD images (6.08, 7.5, etc.). 

If the ZIP drive is connected, the computer boots as expected and I am able to use the RaSCSI mounted HD image. 

Verified hardware using multimeter, everything tested out ok.

edrowley951, 2021-12-21:
Also worth mentioning, i have tried other power supplies and I get the following error in the RaSCSI log:
Error status: Sense Key $05, ASC $24, ASCQ $00

uweseimet, 2021-12-21:
@edrowley951 Does it make a difference if the ZIP is switched on or off? If just the presence of a switched off drive makes a different, a SCSI terminiation issue is likely.

edrowley951, 2021-12-21:
It will not boot (floppy question mark icon) with the zip drive turned off.  As soon as I turn the zip drive on, it boots as expected.

rdmark, 2021-12-22:
@edrowley951 How have you configured termination on the RaSCSI board? There should be two little switches that are either both off, or both on.

edrowley951, 2021-12-22:
They are both on.  I have tried with them off as well, the machine won't even attempt to boot at that point.  I'm going to try a different cable today to see how that goes.  Will comment back once I am able to

edrowley951, 2021-12-22:
Ok, quick update.  It was the cable.  I originally purchased one of the amazon cables linked in the Wiki (6ft version)  I don't know if it was too long or just defective but either way, it absolutely failed with that cable AND being the only device on the SCSI bus.  I purchased one of the 3ft monoprice cables and everything is working as expected. 

Moral of the story, stay away from the Amazon cables!

closed 2021-12-22 by edrowley951

rdmark, 2021-12-22:
Good to hear that you figured out your problem! Happy RaSCSI'ing :)

edrowley951, 2021-12-23:
Thanks for your help!  Glad it was the cable!