PiSCSI/piscsi #559: Finalize localization of rascsi error messages

uweseimet, 2021-12-23:
A few rascsi error messages still do not use the localization feature.

rdmark, 2022-08-28:
@uweseimet Is there anything left to do here?

uweseimet, 2022-08-28:
@rdmark In some cases ReturnLocalizedError() is not yet called. This has to be updated, and localizations added.

n4al, 2022-09-07:
At least one of the (ReturnStatus) messages in src/raspberrypi/rascsi_image.cpp is printed out in the Web UI. This is an issue in case the language of the Web UI is not English. For example:

[image]

uweseimet, 2022-09-21:
@rdmark rascsi.cpp now uses the localizer for all messages. This was the goal of this ticket. If you would like to add other languages than English or German for the new localized messages please create a new ticket.

closed 2022-09-21 by uweseimet