PiSCSI/piscsi #561: Logs flooded with "[warning] accept() failed" when token auth is enabled

rdmark, 2021-12-24:
Under certain circumstances when RaSCSI is running with token auth enabled, it gets stuck in a state where it's spitting out hundreds of these messages every second, freezing other processes running on the Pi, until systemd kills it.

```Dec 21 11:20:48 rascsi3b RASCSI[5645]: [2021-12-21 11:20:44.354] [warning] accept() failed```

I think it happens when a client repeatedly tries to authenticate with either no token, or the wrong token.

uweseimet, 2021-12-27:
@rdmark This error is a socket-related one, I don't think it has anything to do with the authentication, because the authentication is not a low-level (socket-related) feature. It´s part of a message like the device ID etc. and has no special (technical) meaning.

rdmark, 2022-01-09:
I have yet to reproduce this again, and noone in the community has complained. Closing for now!

closed 2022-01-09 by rdmark