PiSCSI/piscsi #565: Named harddisk Connor_CP3200 cannot be attached

bzeiss, 2021-12-26:
# Info

- Which version of Pi are you using: pi4
- Which github revision of software: rdmark_webui_babel
- Which board version: 2.4
- Which computer is the RaSCSI connected to: Yamaha A4000

# Describe the issue

1. In the web-ui, create named disk image
2. Conner CP3200, Create
3. Mount Connor harddisk in the webui
4. 
```
Failed to attach Conner_CP3200.hds to SCSI ID 3 LUN 0
Revision '3.53233' must be between 1 and 4 characters
```

Properties:
```
vendor: CONNER
product: CP3200
revision: 3.53233
block_size: 512
```

akuker, 2021-12-26:
The SCSI `INQUIRY` command will only allow up to 4 characters, which is why this field is limited by RaSCSI. 

>The product revision level field contains four bytes of ASCII data as defined by the vendor. The data shall be left-aligned within this field.

See the "Product Revision Level" in bytes 32-35. 

```
+=====-========-========-========-========-========-========-========-========+ 
|  Bit|   7    |   6    |   5    |   4    |   3    |   2    |   1    |   0    |
|Byte |        |        |        |        |        |        |        |        |
|=====+==========================+============================================|
| 0   | Peripheral qualifier     |           Peripheral device type           |
|-----+-----------------------------------------------------------------------|
| 1   |  RMB   |                  Device-type modifier                        |
|-----+-----------------------------------------------------------------------|
| 2   |   ISO version   |       ECMA version       |  ANSI-approved version   |
|-----+-----------------+-----------------------------------------------------|
| 3   |  AENC  | TrmIOP |     Reserved    |         Response data format      |
|-----+-----------------------------------------------------------------------|
| 4   |                           Additional length (n-4)                     |
|-----+-----------------------------------------------------------------------|
| 5   |                           Reserved                                    |
|-----+-----------------------------------------------------------------------|
| 6   |                           Reserved                                    |
|-----+-----------------------------------------------------------------------|
| 7   | RelAdr | WBus32 | WBus16 |  Sync  | Linked |Reserved| CmdQue | SftRe  |
|-----+-----------------------------------------------------------------------|
| 8   | (MSB)                                                                 |
|- - -+---                        Vendor identification                    ---|
| 15  |                                                                 (LSB) |
|-----+-----------------------------------------------------------------------|
| 16  | (MSB)                                                                 |
|- - -+---                        Product identification                   ---|
| 31  |                                                                 (LSB) |
|-----+-----------------------------------------------------------------------|
| 32  | (MSB)                                                                 |
|- - -+---                        Product revision level                   ---|
| 35  |                                                                 (LSB) |
|-----+-----------------------------------------------------------------------|
| 36  |                                                                       |
|- - -+---                        Vendor-specific                          ---|
| 55  |                                                                       |
|-----+-----------------------------------------------------------------------|
```
_Source:https://www.staff.uni-mainz.de/tacke/scsi/SCSI2-08.html_

Are the additional characters of the revision data being used for something in another SCSI command?

bzeiss, 2021-12-26:
No idea, I just stumbled across this unusable named drive when I tested the i18n feature and wildly created named drives I have never created before. If the revision is limited by the scsi standard, I suspect that the revision for that named drive is likely wrong.

bzeiss, 2021-12-26:
It is the odd man out among the drives:
```
root@raspberrypi:/home/pi/RASCSI/src/web# grep revision drive_properties.json
    "revision": "0A18",
    "revision": "0A18",
    "revision": "1D18",
    "revision": "440C",
    "revision": "0568",
    "revision": "0014",
    "revision": "",
    "revision": "0400",
    "revision": "5000",
    "revision": "2000",
    "revision": "427H",
    "revision": "3.53233",
    "revision": "6.0",
    "revision": "DD24",
    "revision": "",
    "revision": "",
    "revision": "0444",
    "revision": "7462",
    "revision": "2815",
    "revision": "0B70",
    "revision": "",
    "revision": "209A",
    "revision": "0F0C",
    "revision": "0019",
    "revision": "0054",
    "revision": "0054",
    "revision": "0054",
    "revision": "D.13",
    "revision": "D.58",
    "revision": "E.16",
    "revision": "PP33",
    "revision": "4AA0",
    "revision": "0283",
    "revision": "3.1a",
    "revision": "",
    "revision": "4.5d",
```

akuker, 2021-12-26:
Ok. I agree with the bug label now :]

This is a problem with the web interface .json file.

akuker, 2021-12-26:
@rdmark  - it looks like you added this entry to the .JSON file. Do you recall where you found this data?

rdmark, 2021-12-26:
@bzeiss Good catch! I was working off of the INQUIRY data provided by rpajarola, which I think is a mixture of extracted data from real hardware, and various datasheets or other technical documentation. There was probably a typo in [this spreadsheet ](https://docs.google.com/spreadsheets/d/1IcSJdt_3nR2aanl2lmLYrMK_Q6zCuSk0JxYNgolqVJg/edit#gid=754416411) that they provided.

The only reference online that I could find for this Conner drive is from: https://marc.info/?l=freebsd-scsi&m=96222578411583
> CONNER CP3200-200mb-3.5 4040

This is quite different from what we have. I wonder if we should just remove the Conner? It was added as a 'typical' drive on old Sun systems, but no system depends on this profile I think.

akuker, 2021-12-26:
Since it won't work at all right now, it doesn't do any good to leave it in there. I propose deleting it. Its easy enough to add back if/when we get the right version information

rdmark, 2021-12-27:
Merged the PR!

closed 2021-12-27 by rdmark

Pacjunk, 2022-01-06:
If it helps, the DEC RZ24 is a Connor CP3200 with modified firmware.