PiSCSI/piscsi #574: Create version 21.12 release tag and build new image

akuker, 2021-12-28:
- Create a new release tag - v21.12.01
- Generate a new raspberry pi OS image
- Create build release notes

akuker, 2021-12-28:
@uweseimet @rdmark @bzeiss - I'm planning on tagging version 21.12 on Wednesday (December 29) afternoon. Let me know if this doesn't work for you guys.

rdmark, 2021-12-28:
I have no concerns with this release date.
It may not be enough time for folks to add another localization, however.

bzeiss, 2021-12-28:
All good for me.

uweseimet, 2021-12-28:
Also fine for me.

akuker, 2021-12-30:
Since there are still PRs trickling in for the localization stuff, I'm going to hold off on doing the release for a few days.

uweseimet, 2021-12-30:
I recommend to follow the usual release procedures and create a release branch in time, which is open for bug fixes only. This guarantees that changes/PRs related to the develop branch do not affect the upcoming release.

closed 2022-01-03 by rdmark