PiSCSI/piscsi #577: Found a chip replacement for the buffers that Mouser has thousands of.

WarlockD, 2021-12-28:
I was looking around for traslator chips and found the one used in the Atomic PI by chance.  Its called the NTS0104 from NXP.  Apparently its used JUST for stuf like this, but mainly for FPGA's or 1.8v ports.

It autosense input and outputs, supports up to 3.6v on one side and 5.5v on the other.  Has up to 50mbps switching and since they are deisgned for thinks like i2c, they work with open collector inputs/outputs.   It says it can clamp up to -50 mA so it could be used right on the bus.

Only catch is each on is 4 bits only so need 9 of them, also the hand soder version is the TSSOP14 so you will need a magnifying glass on this one to do it by hand:

Going to buy a few anyway.  Want to see if I can make a PicoSCSI now.  Sort of in the vein of the BlueSCSI
[NTS0104-1127256.pdf](https://github.com/akuker/RASCSI/files/7785020/NTS0104-1127256.pdf)

akuker, 2021-12-28:
That's exciting! I'm curious to see how this goes.

Are you primarily looking to reduce the BOM cost? Or physical size? Or both?

I have a RaSCSI Zero that I'm testing out that will give you the size reduction

akuker, 2022-01-06:
Merged into #591. These are both investigating an alternative to SN74LS641-DW

closed 2022-01-06 by akuker