PiSCSI/piscsi #588: What will theoretically die on the RaSCSI if I connect TERMPWR to ground?

jackson2k2, 2022-01-01:
This is not a typical issue form, but:

I noticed magic smoke coming out as I had accidentally done that connecting the RaSCSI's DB25 Macintosh plug to a SCSI card's incompatible pinout, and quickly shut down the Pi. After cleaning, it still seems to work...? Writing works, reading works, recognition works?!

I have not really tested to see if the RaSCSI's termination capability still even works, especially since my SCSI card (which did not get affected) already has built-in termination, so the communication chain is fine for just one device.

What fuses, chips, etc. would I theoretically need to replace if I had accidentally connected TERMPWR to ground for fifteen seconds?

rdmark, 2022-01-03:
@jackson2k2 The magic smoke you saw most likely came from the SMD fuse (FUSE1) on the board. You should still be able to use it as a regular SCSI device after the fuse has blown, but it won't work as a SCSI initiator until you replace it.

See a recent discussion thread on Discord: https://discord.com/channels/749261651978748014/754793914300629114/922604680247390228

jackson2k2, 2022-01-08:
Great!

closed 2022-01-08 by jackson2k2