PiSCSI/piscsi #591: Alternative to SN74LS641-1DW

DottiestDuke47, 2022-01-02:
I ordered some Bare PCBs of the rascsi from JLCPCB, and was checking the price and availibility of the other parts. I found that the SN74LS641-1DW is discontinued and very expensive, so went looking for alternatives. My question is, is the 74HCT245 a viable alternative?

akuker, 2022-01-05:
Hi @DottiestDuke47 - There is a discussion about several different transceiver options here: https://github.com/akuker/RASCSI/wiki/Transceiver-Comparison

The 74HCT245 looks like it is designed to sink a maximum of 35mA, while the SCSI spec calls for 48mA.

In practice, the 74HCT245 would _probably_ work. You would be running it outside of its design tolerances, so it will likely shorten the life of the part or you may have unreliable data.

I don't believe the SN74LS641-1 is discontinued. It has been impacted by the chip shortage though. Texas Instruments still has them available:
https://www.ti.com/store/ti/en/p/product/?p=SN74LS641-1DW

I purchase these transceivers in bulk. So if you're having trouble finding them, I can share what I have.

DottiestDuke47, 2022-01-05:
Thanks for taking the time check and reply to my question - I really appreciate it. Once I receive the PCBs, I will try to assemble them with the chips I have, if they do not work, is there a way I can get in touch to purchase so chips off you?

akuker, 2022-01-06:
Sure. Feel free to email me. My email is the same user name at gmail.com.

akuker, 2022-01-06:
Placing this issue "on hold" until @DottiestDuke47 is able to check out their new PCB design.

akuker, 2022-01-06:
Merging Issue #577 into this issue.

From @WarlockD

> I was looking around for traslator chips and found the one used in the Atomic PI by chance. Its called the NTS0104 from NXP. Apparently its used JUST for stuf like this, but mainly for FPGA's or 1.8v ports.
> 
> It autosense input and outputs, supports up to 3.6v on one side and 5.5v on the other. Has up to 50mbps switching and since they are deisgned for thinks like i2c, they work with open collector inputs/outputs. It says it can clamp up to -50 mA so it could be used right on the bus.
> 
> Only catch is each on is 4 bits only so need 9 of them, also the hand soder version is the TSSOP14 so you will need a magnifying glass on this one to do it by hand:
> 
> Going to buy a few anyway. Want to see if I can make a PicoSCSI now. Sort of in the vein of the BlueSCSI
> NTS0104-1127256.pdf

akuker, 2022-01-06:
Comment from @marciot on Issue #411 :

> This is very tangential to the question at hand, but I am also rather curious whether a bidirectional MOSFET based level shifter such as the ones described in this page could work for SCSI:
> 
> https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all
> 
> Obviously this would require an entirely new board design, and many, many little SMD components, but it would be amazing to have a sort of "universal" Raspberry Pi adapter that could interface with all sorts of retro things, not just SCSI.

DottiestDuke47, 2022-01-19:
So i received the boards yesterday and populated them with the 74HCT245D transceivers, and to my amazement it worked. I tested with a Macintosh SE for about ~45 minutes, everything functioned fine. I'm not too cure about the long term effect of overrunning this chip, but since this is my first vintage mac, I'll be spending a lot of time exploring using this RaSCSI so I can report if anything out of the ordinary happens.[image]

DottiestDuke47, 2022-01-19:
Once again thanks a lot for your help @akuker [image]

DottiestDuke47, 2022-01-19:
> Comment from @marciot on Issue #411 :

With regards to this, how would you control the actual direction of data transfer? On the original raSCSI project, there was a method where you could directly attach a scsi cable to the pi, but the logic levels could fry it, this leads me to belive that we dont need control of A-B B-A but just hi and low level data transfer.

akuker, 2022-01-30:
> So i received the boards yesterday and populated them with the 74HCT245D transceivers, and to my amazement it worked. I tested with a Macintosh SE for about ~45 minutes, everything functioned fine. 

Thank you for reporting back @DottiestDuke47 - Do the chips get very hot while you're using them?

DottiestDuke47, 2022-02-02:
They did get warm but I wouldn't say very hot.

akuker, 2022-02-12:
I spent some time working on this page tonight:
https://github.com/akuker/RASCSI/wiki/Transceiver-Comparison

SN74ABT245B does look like a potential alternative!

I'm going to give this one a try: [SN74F245DW](https://www.ti.com/lit/ds/symlink/sn74f245.pdf?)

thbreidenbach, 2022-02-12:
The abt drivers are among the fastest I know. I wonder what actually limits the speed of the rascsi implementation. But alas the abt drivers are available and since I have a few here I’ll give it a go and report back.

Thanks for the great work!!

Yours Thorsten

akuker, 2022-02-12:
I believe the main limitation with RaSCSI is the speed of the GPIO access. The non-determinism of the Linux kernel doesn't help either. In this way, something like BlueSCSI or SCSI2SD have an advantage. They don't have the overhead of an operating system. 

But, at the same time, they don't have the BENEFIT of an operating system either. Without an OS, things like the Daynaport emulation and the web interface would be much more difficult to implement.

thbreidenbach, 2022-02-12:
I tried the Baremetal implementation on a raspi 3A+ in a fullspec RaSCSI which did 1.3MB/sec on an Amiga 500 with Oktagon 508 with Oktapussy ROM and a reasonable fast Accelerator (35Mhz 68ec020)..
I might swap in an ***@***.*** to see if the Computer limits the speed.

The Zorro 2 Interface of the Amiga should offer something around 3.5MB/sec throughput. A Zorro 3 SCSI Card will be available soon since the A4091 Fast-SCSI 2 Controller is in the process of being reverse engineered. This Card should offer Speeds beyond 16MB/sec.

WarlockD, 2022-03-03:
Ugh I got to check my notifications more often,  Its been a fun few months:P  Anyway the NTS0104 seem to work fine using the pi, but I didn't have time to do any kind of board mock up, but the showed a nice edge on both the 5v and 3 volt size.  That being said, it was a month or so ago and need to rebuild my experiment.

Honestly, the hard IRQ kill that RaSCSI does is probery the best idea.  Its not clean by any means, but there isn't alot of ways you can turn off the scheduler in linux without replacing it completely.  The best way would be to make this a kernel module.  This way it can dirrectly fiddle with the kernel during transfers.

To be honest though, I kind of had to switch to using an ncr53c80.  I am trying to emulate a old tape drive and I HAVE to handle reconnection correctly and as a plus it handles the REQ/ACK easy using the dma pins.

Honestly I kind of wish those old 5V tolernt FPGA's were still around:P

WarlockD, 2022-03-10:
Oh found another.  Seems like it supports auto direction detection, 8-bit port.  NTS0308E.  Push pull is 20mbps but open drain is only 2mbps.  They aren't to bad at 1k@ US$0.54.  Humm, might be fine, max properation delay is 20ns humm, I think the 2 mbps is only in regards to lower votage trasistions and might be fine at 3.3v.  Ah well, I am going to buy a bunch a do some tests.

As for fast scsi-2, its possible, but may not be?  These older controllers usually fudge quite a bit on sync and give way more leway than a more modern controller.  If you kill the irq to the kernel so only your function is running, I think the delay is around 100ns so it MAY be possible as it only matters on the data transmission.  

OH now THIS is intresting.  NVT2010PW is a TSSOP-24, but its a 10bit auto detecting votage between 1-5V.  goes up to 33mhz.  Going to get a few of these as well.

akuker, 2022-03-14:
NVT2010PW does look interesting....

akuker, 2022-03-17:
More data!!!! 

I did some testing with the SN74F245, SN74ALS245A-1 and SN74LS641-1 (as a baseline).

| Device | Max Read | Max Write | Cost each |
| ------- | ---------- | -------------| ------------|
| SN74LS641-1 (baseline) |  950 KB/sec | 506KB/sec | [$3.66](https://www.digikey.com/en/products/detail/texas-instruments/SN74LS641-1DW/1590740)
| SN74F245 | 1245 KB/sec | 574 KB/sec | [$0.93](https://www.digikey.com/en/products/detail/texas-instruments/SN74F245DWR/276824) |
| SN74ALS245A-1 | 1193 KB/sec | 571 KB/sec | [$1.49](https://www.digikey.com/en/products/detail/texas-instruments/SN74ALS245A-1DW/13479511) |

Here are the waveforms of the D0 data line during a read operation:
**SN74F245**
Max Read: 1245 KB/sec
Max Write: 574 KB/sec
[image]
**SN74ALS245A-1**
Max Read: 1193 KB/sec
Max Write: 571 KB/sec
[image]
**SN74LS641-1**
Max Read: 950 KB/sec
Max Write:
[image]

Tests were done with a Raspberry Pi 4, 8GB RAM, running a 22.03 development build. RaSCSI version 2.4a and Digilent Analog Discovery 2 for the O-Scope. The Test PC was a Macintosh SE/30 using SCSI Director 4.0.

akuker, 2022-03-17:
From this data, I'm really leaning towards the SN74F245. The cost can not be beat. Especially since its available from JLCPCB as an assembly option. I'm still going to do more testing, and I'm open to other opinions!

uweseimet, 2022-08-26:
Any news on this?

akuker, 2022-08-26:
Yes, the new boards are sitting on my desk. I just need to finish testing them.

uweseimet, 2022-08-26:
Good news!

thbreidenbach, 2022-10-11:
> FYI: I ordered a FLIR camera so that I can take some temperature measurements between the SN74F245 and SN74LS641-1. This has been my one remaining concern with the SN74F245. I'll report back when I have more data!

I have ordered some older production date 74abt245 (100 I believe) 
Would you like to check those too? They’re quite readily available (at least from me..)

Thorsten

jameswood, 2023-08-09:
> FYI: I ordered a FLIR camera so that I can take some temperature measurements between the SN74F245 and SN74LS641-1. This has been my one remaining concern with the SN74F245. I'll report back when I have more data!

How did you get on with this? I'm using the `SN74F245DW` and they run *hot!* Did you find the same? (I may also have some other problem, of course)

Is the recommendation still to use the `SN74LS641-1DW`?

akuker, 2023-10-14:
I'm closing this ticket, since I'm no longer able to spend time on it. The current recommendation is still to use the SN74LS641-1. The heat issues go away when supplying the SN74LS641-1 with a VCC of 3.3v instead of 5v. However, this is running the chip outside of its advertised limits. I haven't had any issues with it though.

So, in summary, SN74LS641-1DW is still the recommendation.

closed 2023-10-14 by akuker