PiSCSI/piscsi #595: add feature to shut down the pi or put it into a sleep mode when spin down command received

twinec, 2022-01-05:
# Info about my suggestion

Many hard drives completely shut down when the "spin down" command is received over SCSI, so why not have the pi do the same? I'm sure it would be possible to have the pi either shut down or go into sleep mode when the "spin down" command is received over the SCSI bus, and maybe that can be configured in the RaSCSI web interface or the OLED display, depending on the configuration.

akuker, 2022-01-05:
@twinec - thank you for the suggestion!

One difference is that a real hard disk will still have is controller circuitry active so that it can be re-started back up. Shutting down the Raspberry Pi will also shut down the RaSCSI service. This means that there won't be a way for the host to re-start the emulated drives.

twinec, 2022-01-05:
> Do you have any thoughts about how the Raspberry Pi could be re-started after it has "spun down"?

Maybe the shut down would only be used if the RaSCSI is inside a Macintosh Classic, like in my case, but sleep mode would be used otherwise, like if it's in a Powerbook or external enclosure

akuker, 2022-01-05:
Have you observed if the Mac OS sends the "spin down" command when it reboots? Or, does it only send the command when it shuts down? I'm assuming this command is actually the START STOP UNIT command [(1Bh)](https://www.staff.uni-mainz.de/tacke/scsi/SCSI2-09.html)

We're looking at adding supercapacitors to the next revision of RaSCSI to support this case. If power is removed from the RaSCSI, it will hold up the power long enough to save the data to the disk and shut down. But, that development isn't high on the priority list right now.

twinec, 2022-01-05:
> Have you observed if the Mac OS sends the "spin down" command when it reboots? Or, does it only send the command when it shuts down?
I have not observed the behavior of the scsi port when a Macintosh shuts down, because I’m more of a hardware repair guy, but I assume it does that, especially in the powerbooks and macs with soft power capability, but I wouldn’t put it against apple to have the macs without soft power do the same thing. Maybe someone else can observe if the Mac does this.

uweseimet, 2022-01-05:
From what I can tell Mac OS does not send a STOP UNIT command (this is what you mean with spin down I guess). At least I cannot remember having seen this command in any log.
There are platforms which use STOP UNIT to *temporarily* stop drives in order to reduce the noise of drives not in use, just to automatically start the drive again when needed. Shutting off RaSCSI would cause issues because a drive restart would not work.
I tend to recommend rejecting this issue because it causes compatibility problems, and because (assuming that my observation that Mac OS does not send STOP UNIT is correct) it would not have the desired effect anyway.

caver01, 2022-01-06:
Not sure such a command would show up in a log—at least not no the local machine.

uweseimet, 2022-01-06:
It would, see disk.cpp:

```
bool Disk::StartStop(const DWORD *cdb)
{
	bool start = cdb[4] & 0x01;
	bool load = cdb[4] & 0x02;

	if (load) {
		LOGTRACE("%s", start ? "Loading medium" : "Ejecting medium");
	}
	else {
		LOGTRACE("%s", start ? "Starting unit" : "Stopping unit");

		SetStopped(!start);
	}

	// Look at the eject bit and eject if necessary
	if (load && !start) {
		if (IsLocked()) {
			// Cannot be ejected because it is locked
			SetStatusCode(STATUS_PREVENT);
			return false;
		}

		// Eject
		return Eject(false);
	}

	return true;
}
```
If nothing is logged on trace logel STOP UNIT is not sent.

uweseimet, 2022-01-06:
There is one more scenario where implementing this ticket would cause severe issues: A multi-initiator environment. If there is more than one computer (initiator) in the SCSI chain, which is perfectly legal and explicitly supported by SCSI (bus arbitration), you run into severe troubles: If one computer could shut down RaSCSI or the Pi with STOP UNIT, the devices used by the second computer would suddenly be gone.

uweseimet, 2022-01-06:
Closed this ticket because it would cause numerous issues, as outlined in the comments.

closed 2022-01-06 by uweseimet