PiSCSI/piscsi #602: Get supported device types from rascsi rather than hard coded help text

rdmark, 2022-01-11:
The web interface currently has hard coded help text under Image File Management for mapping device type acronyms to human readable device names. Ideally, this should be provided by the rascsi backend instead, to avoid f.e. playing catch-up when rascsi capabilities change.

rdmark, 2022-01-11:
@uweseimet What do you say about introducing something like an enum PbDeviceName with human readable device type names, which is then exposed through the PbDeviceTypeProperties message?

uweseimet, 2022-01-11:
@rdmark Device type names are a matter of localization and of what a client wants to use as a name. Adding localization information to the protobuf interface would not be a good idea, because this is a technical interface. A client can map the PbDeviceType enum values to the translations it needs.

rdmark, 2022-01-11:
@uweseimet Doesn't this mean that each client will have to maintain their own set of canonical device names, that may then diverge between clients? I was just hoping that since supported devices are constant as far as the clients are concerned, that the rascsi backend would be responsible for the canonical human-readable names.

uweseimet, 2022-01-11:
@rdmark Yes, each client is responsible for localizing most of its UI, and the strings used for localization in different clients can diverge in any case. That's the way it is, not just with RaSCSI ;-). Country or language specific data are never part of an interface, and I doubt that you will find any interface that handles this differently.

uweseimet, 2022-01-11:
@rdmark Can it be that your suggestion would fail, not because of (as I tried to point out) wrong use of interfaces, but because you cannot have any human readable string based on enums in an interface definition? You would at least have to escape special characters which are not allowed in enums, like blanks or umlauts. In addition, enum values are by convention upper case, so you would have to set up rules on how to deal with upper and lower case. There are probably more issues of that kind, which besides the technical implications show that interfaces are not meant to deal with localization.

rdmark, 2022-02-07:
Implementation for the Web UI: https://github.com/akuker/RASCSI/pull/655

closed 2022-02-22 by rdmark