PiSCSI/piscsi #603: SCSI Error while formatting disk using Yamaha AW2816 disk utility

n4al, 2022-01-12:
# Info

- Which version of Pi are you using: Raspberry Pi Zero W Rev 1.1
- Which github revision of software: master branch
- Which board version:68mkla 2.4a
- Which hardware sampler is the RaSCSI connected to: Yamaha RS7000 and AW2816

# Describe the issue

Yamaha AW2816 is a hard disk recorder with SCSI connectivity for backups. It comes with a tool to format the attached disk, however when it is executed it only reports "SCSI Error". Being a black box, it is hard to say what the actual problem is. RaSCSI logs indicate 05 24h/00h (INVALID FIELD IN CDB) being the problem. I did not manage to establish more details about this issues. I tried changing the termination on RaSCSI but it did not affect it.

On the other hand, RS7000 (audio workstation) also offers SCSI connectivity to store and load samples, patterns, etc. Without looking at RaSCSI logs, the operation seems correct. It is able to reckonise and format RaSCSI disks, and also to save and load the data that is stored there (by RS7000 itself). However, when looking at the logs, one can spot that 24/00 keeps appearing for some reason.

I have looked at #504 but the timeouts in the master seem to be correct and they seem to match Yamaha's SCSI implementation as described in the ticket.

The steps I used to reproduce this issue on AW2816 are:

1. create a generic 500 mb HDD
2. attach the disk
3. switch on the device
4. format
5. error is observed

The steps for RS7000:

1. use the same disk from the previous set of steps
2. switch on the device
3. format the disk
4. mount the disk
4. load a file from internal memory and save it to SCSI disk
5. load the same file form SCSI disk

I am attaching the trace files that correspond with the steps outlined above. Due to their length, I divided them as follows:

- 1-aw2816-format.txt - trace information when formatting is attempted
- 1-rs7000-boot.txt - trace information during the boot process of RS7000
- 2-rs7000-format.txt - trace information during formatting
- 3-rs7000-write.txt - trace information during writing a file to SCSI disk
- 4-rs7000-load.txt - trace information during loading the saved file
[2-rs7000-format.txt](https://github.com/akuker/RASCSI/files/7855218/2-rs7000-format.txt)
[3-rs7000-write.txt](https://github.com/akuker/RASCSI/files/7855219/3-rs7000-write.txt)

[1-aw2816-format.txt](https://github.com/akuker/RASCSI/files/7855214/1-aw2816-format.txt)
[1-rs7000-boot.txt](https://github.com/akuker/RASCSI/files/7855215/1-rs7000-boot.txt)
[4-rs7000-load.txt](https://github.com/akuker/RASCSI/files/7855221/4-rs7000-load.txt)

Do let me know if I can do something else to proceed with troubleshooting, or if any other information is needed, that I might have missed.

uweseimet, 2022-08-27:
@n4al Can you please re-test formatting with the current release 22.08.02 or the develop branch? Your first log shows that code page $01 was rejected as not being supported. With the current code this code page should definitely be returned.

n4al, 2022-09-07:
I retested it with 22.9.1 35bbe97 and the original problem has been solved. The device reports that the format has been successful. I was able to confirm that this is indeed so by mounting the image locally.

uweseimet, 2022-09-07:
@n4al That's good news! I am going to close this ticket, assuming you agree with that.