PiSCSI/piscsi #605: Works on my VAXStation 4000 model 60!

jeroenbrons, 2022-01-13:
# Info

- Which version of Pi are you using:
- Pi 3B
- Which github revision of software:
- 21.12.-1 c19c814
- Which board version:
- the one from SamplerSpa (2,4 if i'm memory serves right
- Which computer is the RaSCSI connected to:
DEC VAXStation 4000/60

# Describe the issue
Just wanted to notify that the RASCSI works fine on my DEC VAXStation 4000/60 
tested with installing and booting OpenVMS 7.3

Pacjunk, 2022-01-15:
As expected, but good to confirm!

uweseimet, 2022-01-15:
@jeroenbrons Please add your hardware to the RaSCSI compatibility list, so that we can close this ticket.

rdmark, 2022-01-16:
Added to wiki https://github.com/akuker/RASCSI/wiki/Compatibility#System_compatibility

closed 2022-01-16 by rdmark