PiSCSI/piscsi #610: Cant create a 10MB , IMAGE.HDS with Block Size: 256 bytes

CBMD9090, 2022-01-15:
# Info
- Which version of Pi are you using:  Pi 3B+ with Raspberry Pi OS Lite (image from Github)
- Which github revision of software: 21.11.1
- Which board version: Full spec
- Which computer is the RaSCSI connected to: Commodore D9090 Hard Drive - SASI Controller.
- D9090 Hard Drive info:  SASI Controller connected to a TM603S 10MB MFM Hard Drive. Its DOS only Formats about 7.5MB of the 10MB. It uses the following fixed specs:
- Cylinders 153 (0-152), Heads = 6 (or can be set to 4)
- Sectors per Cylinder/Track = 32 , Bytes per Sector = 256
153 x 32 x 256 =29376 Blocks Free (Commodore DOS Blocks)  BAM/FAT steals 214 Blocks, so the user is left with 29,162 Blocks free on a Newly Formatted Disk.

# Describe the issue
I Created a SASI 10MB , IMAGE.HDF with Block Size: 256 bytes, but it does not work with my hardware. So I tired plugging in a real SCSI Hard Drive and it almost works. So I think this SASI device is a lot closer to SCSI in terms of compatibility.

I Cant create a 10MB , IMAGE.HDS with Block Size: 256 bytes
I Created a 10MB , IMAGE.HDF with Block Size: 256 bytes, and then renamed its Extension to IMAGE.HDS, but when I use the Web interface and click the INFO button, it reports:
Product: SCSI HD 10 MB
Revision: 2111
Block Size: 512 bytes

Is there a way to Force raSCSI to allow 256 bytes SCSI HD Image?

uweseimet, 2022-01-15:
@CBMD9090 No, SCSI drives do not support 256 physical bytes per sector. This is only possible with SASI drives, i.e. .hdf is the correct extension. Renaming from .hdf to .hds will not work.
In case of issues with emulated SASI drives you should add a rascsi log on trace level. This may help somebody who is familiar with SASI drives to further analyze your issue.

uweseimet, 2022-01-30:
@CBMD9090 I would like to remind you to add a log on trace level. Without a log it's very unlikely that somebody will ever be able to work on this ticket.

uweseimet, 2022-08-19:
Closing this ticker because a log is still missing. The user appears to have lost interest, and support for 256 bytes per sector media is out of scope anyway.

closed 2022-08-19 by uweseimet

IgorPetrovich, 2023-10-24:
How to do, what new, empty HDD (.HDS), which i create in \ by use emulators can be visible and usable in editdisk (Disk Explorer)? For work with it...
Need special programs for formating HDD ( .HDS ) before use it or what? (Then which programs?
Why emulators have option "Do logical formatting" only for HDD ( .HDF ), and do not have for HDD ( .HDS ) ?

Any ideas, who have information about it?
How creators of popular in internet archives  " X68000 " ; "X68000V4" - " X68000_V4.HDS ", with many games in it on HDD ( .HDS ) , have sucesfull with thsi task? WHat they used for it?

Btw, editdisk (Disk Explorer) opened their X68000_V4.HDS...
My new created by me in Emulators of Sharp x68000 empty HDD ( .HDS ) - no... :(
Any ideas, information, TIP's, Hints?
Need your help, guys... (  Need your information, TIP's, Hints, Tutorials or Lessons )

I want to know, understand, and use HDD ( .HDS ) for - tests, and because this can be have huge sizes of space (Even in Gbs ), much more than  in .HDF ( Which limited just in 40 Mb. )
But, sies not main reason, just want understand - how use .HDS?!
This main, reason, and can't do this task only by myself need your help, who have knows about it and expirence with work with this...

I just know, whta it is possible ( On example with X68000_V4.HDS by developers of this ), just not know how...
Need your help, guys...

IgorPetrovich, 2023-10-28:
After time i found solved on question 1, about how make HDD ( .HDS )
All sucesfully done by this method, which i was do, and typed full information, and guide \ tutorial for someone else who also will be need this ( In future maybe or right now )...:
What i was do step by step for make HDD ( .HDS )?!
1.: Open Emulator XM6 Pro-68k or xm6_typeg_336_20230523
2.: Chose in options - Tools - Create - Fixed Disc SCSI - HDD ( .HDS ) , with sizes for tests 100 \ 1000 Mb ( In Fact 99 \ 999 Mb)
Choose Sizes ( Capacity )
Other Options:
Options Must Be On "Attached to interface." And "Replace First Drive."
Press "OK"
Checking What In Options Of Emulator, In Option SCSI - SCSI Interface - Internal - SCSI Discs ( via SCSI Interface ) - Drives ( 1 ) - ID ( 0 ) - SCSI-HD Image File ( .HDS )
3.: After created HDD ( .HDS ), i opened Emulator XM6 Pro-68k \ xm6_typeg_336_20230523 , Options - Floppy Drive #0 - MasterDisk_V3
4.: In DOS , in first i typed Command "FORMAT"
5.: Chosed "SCSI Devices" - Enter - "Hard Disc" - Enter - "Initalise" - Enter ( Y ) - "Create" - Enter - "Size"- Enter - Chosed Sizes ( 99 Mb ), by Typed this in window ( In Down Of The Screen ) - Enter -"Start" - Enter - "Create new partition" - "Y" , and got this results in end:
https://i.ibb.co/M2pdx89/44444444.png
6.: "End" - Enter.
7.: Do you want to reset? Most changes to the Partitions need it... Chosed - "Reset" ( By Press "Y" )
Done!
HDD Disc, with format .HDS - already created, formatted, can be readable now, now must be visible( In Programs \ Utilities \ In Emulators ), and ready to work ( For example in Disc Explorer ) .
P.S.:
I really like to write a lot and in detail, describing each point in detail... :)