PiSCSI/piscsi #611: Feature request: HD/Floppy sound

TechGuyVincent, 2022-01-15:
Hi! 

Just for fun (again if possible) - could one have a soundoutput simulating the SCSI HD / Floppy seek etc? 

Thinking of something like the "Noisy Disk Mechanical Sounder" - but software. Again, just for fun. I miss my old SCSI HD noises :D 

/V

marciot, 2022-02-07:
This probably isn't a RaSCSI feature, but an extension for the Macintosh. Back in the days I wrote an extension that would cause a Mac to make a sneeze sound whenever a disk was ejected.

TechGuyVincent, 2022-02-08:
True, but since the RASCSI runs on a Pi I though it might be easy to add a piezo speaker of some kind/small speaker and have some noises generated when the RASCSI does read/write operations. Again, totally unnecessary - but would be a nice touch.

uweseimet, 2022-08-26:
@TechGuyVincent Please do not feel offended when I ask @rdmark and @akuker whether this is a realistic ticket in the scope of rascsi. Most users I know (also of retro computers) prefer *not* to have any drive noises, and thus replaced their drives by SSDs, memory cards ore solutions like rascsi.
And users would really like the noise might want to hear the noises of specific drive models, and we all know they do differ ;-). So this would have to be configurable and so on.
Additionally, if rascsi was to somehow create noises, this might even degrade performance, because it would have to happen on regular read/write operation, when rascsi is busy anyway.

akuker, 2022-08-26:
I believe this could be achieved by attaching a relay mechanism to GPIO4 (the activity LED) or GPIO21 (The SCSI ACK signal). These should be attached on the Raspberry pi side of the transceivers and should have appropriate isolation from the relay.

Using a speaker though, would degrade the performance of RaSCSI. We are extremely limited on GPIO pins as well, so I wouldn't want to dedicate a specific pin for this function.

closed 2022-08-26 by akuker