PiSCSI/piscsi #618: Add support for MODE SELECT

xelalexv, 2022-01-18:
# Info

- Which version of Pi are you using: Pi Zero W
- Which github revision of software: 21.12.1 + fix for issue #614
- Which board version: 68kmla 2.4a
- Which computer is the RaSCSI connected to: SUN SPARCstation IPC (SUN 4/40)

# Describe the issue
I'm attaching a Solaris 2.3 install CD-ROM ISO at id 6. This essentially works, and I can mount the CD from within SunOS. However, when I try booting from the CD, the boot process starts, but after a short while errors out and triggers a reboot from the internal HDD. This was initially caused by a problem with `MODE SENSE` (see issue #614). After applying the provided patch for this, a second issue with `MODE SELECT` has come up, and boot still aborts as before. When this happens, the rascsi logs have this to say:

```
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $1A
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSense6 ($1A)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] int Disk::ModeSense6(const DWORD*, BYTE*) Requesting mode page $01
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] int Disk::ModeSense6(const DWORD*, BYTE*) 24 bytes available, 20 bytes requested
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] void SASIDEV::DataIn() Going into Data-in Phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 20
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] void SASIDEV::Status() Status phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] void SASIDEV::Status() Status Phase $00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.730] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.731] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.731] [trace] void SASIDEV::MsgIn() Starting Message in phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.731] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.731] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.731] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.731] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.731] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] void SCSIDEV::MsgOut() ID 6
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] Message Out Phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] void SCSIDEV::MsgOut() ID 6
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] virtual void SCSIDEV::Receive()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] virtual void SCSIDEV::Receive() length is 1
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] void SCSIDEV::MsgOut() ID 6
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] virtual void SCSIDEV::Receive()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.732] [trace] Message code IDENTIFY $C0, LUN 0 selected
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SASIDEV::Command() Command Phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SASIDEV::Command() CDB[0]=$15
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SASIDEV::Command() CDB[2]=$00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SASIDEV::Command() CDB[4]=$14
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SCSIDEV::Execute() Execution phase command $15
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $15
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $15
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSelect6 ($15)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void Disk::ModeSelect6(SASIDEV*) Unsupported mode page $15
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] void SASIDEV::DataOut() Data out phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SCSIDEV::Receive()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SCSIDEV::Receive() length is 20
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SCSIDEV::Receive()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 7 (dataout)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [debug] Error status: Sense Key $05, ASC $26, ASCQ $00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] void SASIDEV::Status() Status phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.733] [trace] void SASIDEV::Status() Status Phase $02
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] void SASIDEV::MsgIn() Starting Message in phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.734] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] void SCSIDEV::MsgOut() ID 6
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] Message Out Phase
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] void SCSIDEV::MsgOut() ID 6
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] virtual void SCSIDEV::Receive()
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] virtual void SCSIDEV::Receive() length is 1
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] void SCSIDEV::MsgOut() ID 6
[... cut ...]
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] virtual void SASIDEV::Command() CDB[4]=$14
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Jan 18 12:37:44 rascsi RASCSI[351]: [2022-01-18 12:36:49.735] [trace] virtual void SCSIDEV::Execute() Execution phase command $03
```

Apparently, `MODE SELECT` is currently not supported, so seeing support for this added would be terrific!

uweseimet, 2022-01-18:
The requested sub-page, according to the CDB from the log, is $00.

xelalexv, 2022-01-19:
I tested this, but still no boot. I had no time to properly check the RaSCSI logs, but may have found a workaround: Just for the heck of it, I attached the Solaris installer ISO at ID 2 as a hard disk instead of CD-ROM drive, and with that it boots up :laughing:  At least I get to the main menu of the installer. Whether it will go through I don't know yet, may try out later.

uweseimet, 2022-01-19:
@xelalexv Interesting work-around, indeed :).

xelalexv, 2022-01-20:
I did a few more tests with the workaround described above, and with the RaSCSI on the SPARCstation IPC in general. I can confirm that attaching a CD-ROM ISO as a hard disk in RaSCSI allows you to boot from it without any problems. With this, I could successfully do the following:

- boot into single user mode using a Solaris installer CD
- `dd` the internal physical HDD into a virtual HDD emulated by RaSCSI
- swap the SCSI IDs of physical & emulated HDD
- boot from the emulated HDD - system works like a charm, no problems so far

I can pretty much do now what I intended to do with RaSCSI and my IPC, that is back up the original installation and then play around with different versions of Solaris using virtual disks. Pretty cool! Should any of this info maybe go into the supported systems section of the wiki?

uweseimet, 2022-01-30:
@rdmark @xelalexv I suggest to close this ticket. Support for all potential mode pages is not possible anyway, and if a particular mode page is missing in the future, a specific ticket should be opened for such a case.

xelalexv, 2022-02-06:
@uweseimet  @rdmark I've validated the workaround also with a SPARCstation IPX. Everything works, so closing this ticket.

closed 2022-02-06 by xelalexv

RolopoloRolo, 2022-05-21:
I have started to clear enough space to try to get Nextstep installed on a HP715/100 using RaSCSI alone and a SPARCstation 10. The HP is running already with another modern SCSI solution, for that I successfully used Jaz drive with an image of the install media. Don't think the HP is fussed about the identify of the device, more its capabilities. So far lock up once the installer gets to really in to going when it normally would with a CDROM drive it does not like. I don't have enough experience yet with the RaSCSI but is this something you would be interested in me following up. The RaSCSI works fine inside Nextstep as a CDROM emulator. Not tried to boot from an installed OS yet though but so fits in with your notes that it is working.

oconnelltim, 2022-07-23:
Hi folks - I'm just wondering if some more information on this method of booting Sun4c machines can be made available here or somewhere, as I haven't been able to make this workaround work using multiple Sun-4c machines (SS2, SSIPX, etc).  

If I try to mount a Solaris image on SCSI ID 2 as an HD image, I just get "The file loaded does not appear to be executable". I get the same error no matter what version Solaris ISO I use (2.3, 2.4, 2.5.1, etc).  The command I've used to mount the image is `rasctl -i 2 -c attach -u 0 -t hd -b 512 -f solaris_2.5.1_1197.iso`.  Otherwise, if I try to boot these images as CD images using the usual SCSI ID 6, it does start booting the kernel, but then I get:

`WARNING: /sbus@1,f8000000/esp@0,800000/sd@6,0 (sd6):
                     sr_check_blk_size: Mode Sense Failed`

I'm using software version 22.05.2, and my RaSCSI hardware is 2.4a. I have almost all the Sun-4c (except the ELC/SLC) so am happy to test anything out if that would help.

xelalexv, 2022-07-23:
I checked my notes, and this was the command I used to mount the installer ISO:

```
rasctl -c a -i 1 -f /home/pi/images/solaris-2-3.iso -b 512 -t hd
```
So, pretty much like yours, but I didn't use the `-u 0` option.

oconnelltim, 2022-07-24:
Thanks @xelalexv for the quick response.  It turns out that part of this was PEBCAK, and part of it is compatibility of this method with newer Solaris versions/ISO images. 

Your command does indeed work; the correct corresponding Sun boot command for Sun-4c machines with older (v1.x) boot PROMs is:
```
boot sd(0,X,2)
```
Where `X` in the above is the SCSI ID you want to boot the CD image from.  

The correct Sun boot command for Sun-4c machines with newer (v2.x) boot PROMs is:
```
boot /sbus/esp@0,800000/sd@X,0:c
```
Where `X` in the above is the SCSI ID you want to boot the CD image from.

The issue I was having here is that I was forgetting to tell the OBP (Open Boot PROM) to boot from slice 2.  This is automatic in OBP when you use SCSI ID 6, but not when you use any other SCSI ID. 

Using the `rasctl -c a -i 1 -f /home/pi/images/<image_name> -b 512 -t hd` command on the RaSCSI, I had the following results:

SPARCstation 1+ (ROM v1.3) - boot using `boot sd(0,1,2)`: 
- sunos414.iso: success
- Solaris1_1_2.iso: success
- Solaris_2.1_SPARC.iso: success
- solaris_2.2_sparc.iso: success
- solaris_2.3_sparc.iso: success
- solaris_2.4_sparc.iso: success
- solaris_2.5.1_1197.iso: success
- solaris_2.6_598.iso: fails with `panic_vfs_mountroot: cannot mount root`
- solaris_7_1199.iso: fails with `panic_vfs_mountroot: cannot mount root`

SPARCstation 2 (ROM v2.9) - boot using `boot /sbus/esp@0,800000/sd@1,0:c`:
- sunos414.iso: success
- Solaris1_1_2.iso: success
- Solaris_2.1_SPARC.iso: success
- solaris_2.2_sparc.iso: success
- solaris_2.3_sparc.iso: success
- solaris_2.4_sparc.iso: success
- solaris_2.5.1_1197.iso: success
- solaris_2.6_598.iso: fails with `panic_vfs_mountroot: cannot mount root`
- solaris_7_1199.iso: fails with `panic_vfs_mountroot: cannot mount root`

In all cases, SunOS/Solaris complains about the CD image/disk having corrupt label/bad geometry and a label/drive size mismatch, but for the cases where it works, it just continues without any issues. 

However, for the Solaris 2.6 and 7 images which fail above, there is a workaround: 
- Use the `TOSHIBA XM-3401TA` named CDROM disk image in the 'Create Named Drive' dialog in the RaSCSI UI
- Mount it as SCSI Device 6 (the default Sun CDROM SCSI ID)
- Boot v1.x OBP systems with `boot sd(0,6,2)` or boot v2.x OBP systems with `boot cdrom`

Also, for simplicity's sake, for Sun-4c machines running v2.x of OBP where there is an automatic alias between Slice 2 of SCSI ID 6 and the alias `cdrom`, you can also just use this method to mount the image as a SCSI HD on ID 6, i.e.:

On RaSCSI:
```
rasctl -c a -i 6 -f /home/pi/images/sunos414.iso -b 512 -t hd
```

On the v2.x OBP Sun-4c machine:
```
boot cdrom
```

Thank you for your help!

xelalexv, 2022-07-24:
Awesome! Great detailed info! I never went higher than Solaris 2.3 on my IPC & IPX (and probably never will), but great reference should I ever change my mind.