PiSCSI/piscsi #639: RTC processor device extension: non-volatile RAM support

uweseimet, 2022-02-02:
The RTC processor device (https://github.com/akuker/RASCSI/issues/638) shall be extended to support non-volatile RAM (NVRAM) data. An additional vendor specific mode page shall be added:

This is the payload of the custom mode page with the vendor-specific page number 0x21:

```
typedef struct {
  // Major and minor version of this data structure
  uint8_t major_version;
  uint8_t minor_version;
  // Up to 512 bytes of NVRAM data
  // Read-only, at least in the initial implementation
  int8 data[512];
} nvram;
```

Note that these data are located at offset 4 or 8 of the returned mode page data. (See SCSI specification for details on MODE SENSE(6) and MODE SENSE (10).) The data format is big endian, as far as relevant.

RaSCSI shall store the NVRAM data like it stores mass storage data in disk image files, so that the existing code can be reused. The NVRAM data file can be specified when attaching the device, just like it is done with mass storage devices. In case the Disk base class is used (not ideal, but probably required)

An Atari client might implement the NVMaccess() system call for TOS versions without native NVRAM support.

uweseimet, 2022-02-03:
Won't be implemented, because the NVRAM data are only useful if they are available before booting a driver that can read them via SCSI.

closed 2022-02-03 by uweseimet

akuker, 2022-02-04:
Cool idea, but feels like we're implementing a disk that isn't a disk. You could accomplish the same thing with a small drive image.

uweseimet, 2022-02-04:
@akuker This ticket has already been closed/rejected :).