PiSCSI/piscsi #642: Add an general purpose SCSI processor device as rascsi control interface

uweseimet, 2022-02-03:
It should be possible to manage some functionality of the rascsi process with SCSI commands. MODE SELECT/MODE SENSE could be used for that, similar to how it is done for the RTC feature of https://github.com/akuker/RASCSI/issues/638.
SCRT (where RT means realtime) should be renamed to SCRA (where RA means RaSCSI). SCRA would then be the device that (via MODE SENSE and MODE SELECT) makes it possible to manage certain RaSCSI features or other functionality via SCSI commands. The existing realtime clock feature would not change.
A new feature might be a means to shut down RaSCSI or the Pi with MODE SELECT, which would partially address https://github.com/akuker/RASCSI/issues/335. Quoted from https://github.com/akuker/RASCSI/issues/335:

"The scenario of having the Pi powered by the host and switching off the host can be addressed up to a certain degree by implementing a custom SCSI command (or a vendor specific mode page for MODE SELECT) that shuts down RaSCSI or the whole Pi. An OS that can run scripts (or binaries) during its shutdown phase could launch a script that sends this custom command. Linux or any Unix can do that, or MiNT or MagiC for the Atari. I guess a Mac can also execute code during its shutdown.
The actual shutdown process already exists as part of the RaSCSI remote protobuf interface (SHUT_DOWN operation), but currently it can only be triggered by rasctl, the web interface or the RaSCSI Control app. What's missing is a means to do the same on the SCSI level."

Essentially renaming SCRT to SCRA and using a more general INQUIRY device name than "RTC" is a resolution for this ticket, because any new functionality can be added any time with a new mode page.

uweseimet, 2022-02-03:
@akuker @rdmark I would like to hear your opinions on this. Essentially this ticket generalizes the RTC approach. There is no need to restrict RaSCSI to only emulate features of existing (old) SCSI devices. Platforms that have official interfaces for sending arbitrary SCSI commands (Linux or the Atari, for instance) can use SCSI commands as a general interface to rascsi or the Pi.

akuker, 2022-02-04:
@uweseimet - I really like this idea. It opens up the door for someone to make a controlling app for the connected system. 

In the RT issue, you talked about sharing a LUN. I'm assuming this isn't a requirement and this device could live on its own device ID (just like anything else).

uweseimet, 2022-02-04:
@akuker Sure, it is a regular device, just like any other. All SCSI devices represent a LUN, usually LUN 0. If you have drivers that support multiple LUNs the nice thing is that you can attach RaSCSI devices without using up a valuable SCSI ID for each of them. But again, this is not new functionality.