PiSCSI/piscsi #648: SCRA: Trigger script/binary execution on the Pi with a SCSI command

uweseimet, 2022-02-06:
The RaSCSI service device (SCRA) shall be extended to execute code on the Raspberry Pi. MODE SELECT with a vendor-specific mode page might be a candidate for the required SCSI command.

Requirements:

- The command and its arguments have to be passed
- The UID (and GID?) to use for the executed command have to be provided. Default is 0 (root), which is not a security risk, because there is no way to trigger command execution by any other means than a computer connected to the same SCSI bus
- The exit code of the command shall be returned. There may be commands that do not terminate. This is why oneway semantics (equivalent to starting a command in the background) shall be supported

akuker, 2022-02-07:
> * Default is 0 (root), which is not a security risk, because there is no way to trigger command execution by any other means than a computer connected to the same SCSI bus.

I agree, this isn't really a security risk. 

Do you have any example use cases for this feature?

An interesting extension to this would be to access the 'rasctl'-like functions. One could develop a host application that would do the same functionality as the android/web apps, but going through the scsi bus.

uweseimet, 2022-02-07:
@akuker No, no examples yet from my side. This ticket is in an early stage. If nobody finds any example use case it should not be implemented. Typical examples would be cases where the Pi can do work for SCSI clients.

uweseimet, 2022-08-19:
Still no ideas on meaningful binaries/scripts that could be executed, so I am going to close this ticket.

closed 2022-08-19 by uweseimet