PiSCSI/piscsi #652: Segfault when attaching SCBR after another network device

rdmark, 2022-02-07:
On the latest develop:

If you attach a SCBR after another network device (either SCBR or SCDP) are already attached, rascsi crashes with a segmentation fault. While this is not exactly a valid usecase, I'd expect it throwing a helpful error rather than crashing.

```
dmark@rascsi3b:~/dev/RASCSI/src/raspberrypi $ sudo rascsi &
[1] 7597
dmark@rascsi3b:~/dev/RASCSI/src/raspberrypi $ SCSI Target Emulator RaSCSI(*^..^*) version 22.03 --DEVELOPMENT BUILD-- (Feb  6 2022, 21:52:56)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2021 Contributors to the RaSCSI project
Connect type : FULLSPEC
[2022-02-06 22:10:09.898] [info] Set log level to 'info'
[2022-02-06 22:10:09.901] [info] No images currently attached.
No images currently attached.
dmark@rascsi3b:~/dev/RASCSI/src/raspberrypi $ rasctl -i 0 -c a -t scbr
[2022-02-06 22:10:22.791] [info] Validating: operation=ATTACH, command params='locale=en_GB.UTF-8', device id=0, unit=0, type=SCBR, vendor='', product='', revision='', block size=0
[2022-02-06 22:10:22.792] [info] Executing: operation=ATTACH, command params='locale=en_GB.UTF-8', device id=0, unit=0, type=SCBR, vendor='', product='', revision='', block size=0
[2022-02-06 22:10:22.798] [info] rascsi_bridge is already available
[2022-02-06 22:10:22.800] [info] Tap device ras0 created
[2022-02-06 22:10:22.801] [info] Attached SCBR device, ID 0, unit 0
dmark@rascsi3b:~/dev/RASCSI/src/raspberrypi $ rasctl -i 1 -c a -t scbr
[2022-02-06 22:10:27.309] [info] Validating: operation=ATTACH, command params='locale=en_GB.UTF-8', device id=1, unit=0, type=SCBR, vendor='', product='', revision='', block size=0
[2022-02-06 22:10:27.312] [info] Executing: operation=ATTACH, command params='locale=en_GB.UTF-8', device id=1, unit=0, type=SCBR, vendor='', product='', revision='', block size=0
[2022-02-06 22:10:27.315] [error] Error: can't ioctl TUNSETIFF. Errno: 16 Device or resource busy
Error:
dmark@rascsi3b:~/dev/RASCSI/src/raspberrypi $
[1]+  Segmentation fault      sudo rascsi
```

This is what happens if you try to attach an SCDP device instead at the last step:
```
dmark@rascsi3b:~/dev/RASCSI/src/raspberrypi $ rasctl -i 0 -c a -t scdp
[2022-02-06 22:14:18.633] [info] Validating: operation=ATTACH, command params='locale=en_GB.UTF-8', device id=0, unit=0, type=SCDP, vendor='', product='', revision='', block size=0
[2022-02-06 22:14:18.634] [info] Executing: operation=ATTACH, command params='locale=en_GB.UTF-8', device id=0, unit=0, type=SCDP, vendor='', product='', revision='', block size=0
[2022-02-06 22:14:18.634] [error] Error: can't ioctl TUNSETIFF. Errno: 16 Device or resource busy
[2022-02-06 22:14:18.634] [error] Unable to open the TAP interface
[2022-02-06 22:14:18.636] [error] Initialization of
Error: Initialization of  device, ID 0, unit 0 failed
```

No segfault! :)

uweseimet, 2022-02-07:
@rdmark Should be fixed with https://github.com/akuker/RASCSI/pull/654. Must have been wrong since the 21.10 release or so.

rdmark, 2022-02-08:
@uweseimet Thanks for the quick fix!

closed 2022-02-08 by rdmark