PiSCSI/piscsi #656: Mac plus write error

edgeux, 2022-02-07:
- I have tried a Pi 4/4gb and a 3B+ 
- both Jan release and feb release fullspec
- 2.4a 68klma
-  Mac plus 4gb

# Describe the issue
I have tried all of the boot images supplied in the repo as well as manually created system (6.0.8, 7.0.1, 7.5.3)
I have tried booting from USB and from SD
I have tried both 32 and 64 bit rpi os
I have tried multiple scsi cables. 
Voltage is ok.

I am able to boot from all/any rascsi system to a functional desktop HOWEVER when i try to perform any operation that requires writing to the scsi disk the mac freezes after ~10 seconds of data transfer. 

with the log set to TRACE, the rascsi/macplus hangs reliably on these two lines:

```
Feb  7 19:45:31 max RASCSI[484]: [2022-02-07 19:45:31.429] [error] virtual void SCSIDEV::Receive() Not able to receive 512 data, only received 2. Going to error
Feb  7 19:45:31 max RASCSI[484]: [2022-02-07 19:45:31.429] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
```

the only difference with each hang being that:

`only received 2
`
varies numerically  in each case.

i am able to boot from a system floppy with zero errors but as soon as i write to scsi, i get maybe 10 seconds of transfer before hanging with the same errors

advice, instructions, etc ?

m

rdmark, 2022-02-08:
@edgeux Does other SCSI storage devices work well with your Plus? Have you tried using contact cleaner on the SCSI port on the computer?

edgeux, 2022-02-08:
Thanks for the reply, I have performed contact tests with every pin board to board. I have no other scsi devices to test unfortunately.

uweseimet, 2022-02-08:
@edgeux Is your SCSI chain properly terminated? The physical ends of the chain have to be terminated, i.e. if the RaSCSI board is your only device it definitely must be terminated.
Considering that your problem are very obvious, but there are no reports of similar errors by error users (at least not tickets), indicates that your problem is very likely related to your specific hardware setup (including termination).

edgeux, 2022-02-08:
It is the only scsi device and the jumpers are terminated on the rascsi. I think I will have to get my hands on some other scsi equipment to test further. Is there any other useful information that can be derived from the error itself? It means nothing to me.

uweseimet, 2022-02-08:
@edgeux I'm afraid there is no further information the error message provides. It just indicates a basic SCSI communication/protocol issue.

supercompman, 2022-02-12:
I'm seeing the exact same error with my Mac Plus. I'm using a Raspberry Pi 3b, RaSCSI board v2.4, 4MB RAM installed on the Mac Plus, and the RaSCSI is the only device connected to the SCSI bus, with termination enabled. The Mac Plus boots fine from floppy, a FloppyEmu (either configured as either a floppy drive or HD20), and from a SCSI Zip drive. The RaSCSI works completely fine on a Mac Classic II.

supercompman, 2022-02-12:
@uweseimet Yes. I'm seeing the same things in my logs. I'll post my logs as soon as I get a chance if that would help.

uweseimet, 2022-02-12:
@supercompman If your trace logs shows do same you don't have to spend on posting it. I just wanted to ensure from the logs your issue is the same, and not just a similar problem with different logs.

uweseimet, 2022-02-25:
I also stumbled upon this error message (with various counts for the received data), but it is not reproducable.

```
[2022-02-25 13:12:23.845] [error] void SCSIDEV::ReceiveBytes() Not able to receive 512 data, only received 446. Going to error
```

supercompman, 2022-07-13:
After experimenting on and off for a few months, I finally found a solution that worked for myself. I have no idea if it will work for anyone else. Using Silverlining 5.31 for the driver and partitioning worked well. Previously, I had been using the Apple tools and driver. With Silverlining, I have the option of selecting the method used by the driver for reading and writing data. Any of the "Blind" methods would not work at all, while all of the "Handshake" methods worked perfectly. I reviewed my RaSCSI logs and I'm not seeing any errors when using the Silverlining driver.

jcres69, 2022-07-30:
@supercompman 
Thank you!  I started using the Silverlining driver a few weeks ago and so far the SCSI errors and freezes on my mac plus have disappeared.

rdmark, 2022-08-19:
I've documented in wiki the fact that only Silverlining drivers are known to work properly with the Plus. https://github.com/akuker/RASCSI/wiki/Drive-Setup#initializing-a-new-disk-on-classic-mac-os

In the long term it'd be great to have a workaround in RaSCSI code, so let's keep this ticket open.

uweseimet, 2022-08-19:
@rdmark It will be very hard to find out what's wrong because these errors are spurious and we do not know how to reproduce them. This might be a timing issue, or even a general (not by board, but by design) hardware issue.

supercompman, 2022-08-20:
@uweseimet On my Mac Plus, the issue is anything but spurious. If there's any information that I can gather or testing I can do to help understand this issue better, just let me know. I'm not a C/C++ or Python developer, but I am a  software developer by trade and I'm pretty comfortable working with Linux.

uweseimet, 2022-08-20:
@supercompman I'm afraid I cannot provide further support in this case. I think that this particular issue needs to be debugged and resolved by somebody who has a setup where this problem happens frequently. Somebody like you ;-). A try-and-error approach may be needed.
it's interesting that even though the Mac Plus is not the only platform where this issue can be observed, it appears to be the platform affected most, while the effect on other platforms is neglectable. Whatever it is, there must be a reason for this.

rdmark, 2022-08-21:
@supercompman I think Uwe's point is that the Receive() error is a red herring, and the actual issue lies elsewhere.

@uweseimet Yes fair point, I can see how the issue is likely not driver specific. We have also observed issues with Lido drivers, RaSCSI, and the Mac Plus in the past. However, using Apple drivers is the sure way to trigger the most severe effect, so it's still worth mentioning as a reproduction path.

As a side note, in order to use a SCSI2SD with the Mac Plus, you need to configure the device with the following compatibility settings:
- SCSI Selection Delay = 0 (default:255)
- Respond to short SCSI selection pulses = True (default:False)

No idea if this helps. I couldn't find the source code for scsi2sd-util v5.x to see what this actually does.

rdmark, 2022-10-24:
To all of you who have had issues on your Mac Pluses: Please try the new 22.10 release. From our testing, the stability on the Plus is much improved in this version. https://github.com/akuker/RASCSI/releases/tag/v22.10.01

rdmark, 2022-11-03:
@uweseimet I would like to revisit this issue. I have a Mac Plus myself now and am able to assist with debugging.

To remove all other factors, I followed these test steps:
- Created blank 20MB disk image with the "Miniscribe" profile in order to HD SC Setup to recognize it.
```
{
    "device_type": "SCHD",
    "vendor": "MINSCRIB",
    "product": "    M8425 - SCSI",
    "revision": "209A",
    "block_size": 512,
    "size": 20994048,
    "name": "Miniscribe M8425",
    "file_type": "hds",
    "description": "Recognized by unpatched Apple HD SC Setup.",
    "url": ""
},
```
- Booted the Plus from System Tools floppy disk (System 6.0.0)
- Launched HD SC Setup v2.0
- Initialized the disk image (this creates the partition map, injects the disk driver, creates a file system, and mounts it)
- Turn on trace level logging in rascsi
- Copied the contents of the System Tools floppy disk (800k worth of data) onto the initialized disk image
- Observe file transfer errors
- With 100% repro rate, a write error will occur for at least one file (sometimes multiple files; it's not deterministic)
- Leading up to the error, the file transfer process will pause for 20-30s before throwing the error in the OS, at which time a huge amount of logs are repeatedly output (the majority of attached logs)

rdmark, 2022-11-03:
[image]
[pluslog.txt](https://github.com/akuker/RASCSI/files/9930981/pluslog.txt)

uweseimet, 2022-11-03:
@rdmark It appears as if the bus signals say that the DATA OUT phase is over before a full sector has been transferred. This is not an issue with the SCSI command implementations, but something that happens during the data transfer. Saying that I am not surprised that despite all of my changes the issue is still there, because my changes do not deal with the low-level signal handling and/or timing. The problem could also be a general RaSCSI hardware issue. If I remember correctly @akuker mentioned that he has some bus signal analyzer hardware. This might be what's needed in order to find out more.
It's amazing that since I also stumbled upon this issue early this year it has never happened again.

rdmark, 2023-05-12:
For the record, if someone reads this trying to get their Plus working, I prepared a modified MacPack image that uses Silverlining drivers and posted it here:

rdmark, 2023-10-22:
@uweseimet Unfortunately I had to give away both my Mac Pluses when I relocated internationally last month.

Is there anyone else subscribing to this thread who would test on their Plus? It'd be valuable to hear if the behavior has changed in any way in the branch that Uwe refers to.

uweseimet, 2023-10-22:
One problem with this ticket appears to be reproducability. I have not seen this error with my setup for ages, but this does not mean that the problem is gone. It would be good if somebody had a setup where this error is quite reproducable, now that @rdmark does not have his hardware anymore.

predecker, 2024-03-03:
I can confirm the issue exists on 
`PiSCSI Software-Version: 23.11.1, Hardware und OS: Raspberry Pi 3 Model B Rev 1.2, Linux 6.1.21-v7+ armv7l` 
actually. 

I downloaded HD20_512-1GB-MacPlus-6+7.hda from somewhere as my 6.0.8 bootdisk had no tools to initialize the disk.
The image booted with 6.0.8, but Silverlining from that image was unhappy to start. So I used HD SC Setup to initialize a 200MB image. When trying to copy the system 7 folder from the image to the new disk I had the mentioned log entries:
`Mar  3 17:40:08 raspi-rascsi PISCSI[457]: [2024-03-03 17:40:08.379] [error] (ID 5) - Not able to receive 512 byte(s) of data, only received 97`
The last number changes with every line.
PiSCSI is the only device on the bus, it ran flawlessly on a Mac Classic last time I used it. That's a year ago however.
Both switches are on the "on" position.
I have an oszilloscope if that helps ...