PiSCSI/piscsi #660: SASI controller has to delegate SCSI command dispatching

uweseimet, 2022-02-09:
The execute() method of the SASI controller shall dispatch commands to the affected device, just like the SCSI controller does it in its execute() method.
The fact that the controller knows how to execute specific commands breaks encapsulation and makes it impossible to clean up the Disk class.

uweseimet, 2022-02-09:
@akuker Is there anybody who can test the result of addressing this ticket? IMO in the ideal case we would remove all the SASI stuff. The implementation is bad and it is only causing troubles and wastes time. Any chance to get rid of it? Is anybody using/testing this at all?

akuker, 2022-02-12:
I brought it up in the Discord chat a while back, and there are quite a few people using the SASI stuff. I agree that it makes the code much more confusing and harder to work with. But, if people are using it, I think we should try to leave it in.

uweseimet, 2022-02-12:
@akuker I agree, if it is still being used it should not be removed. Nevertheless, I would like to improve the code. Is there anybody who is willing to test my changes with the required hardware?

uweseimet, 2022-03-05:
@akuker Any chance to ask somebody from those using SASI to test the changes required for this ticket? If not it's probably best to completely decouple the SCSI from the SASI implementation (this will unfortunately duplicate code), so that the SASI implementation can be frozen and does not block further progress.

uweseimet, 2022-03-05:
@akuker Are you really sure that somebody is using the SASI emulation? I just tried to create a SASI device, and on first access there is a segfault. I am quite sure you can reproduce this with a Mac, just be running "rascsi -hd 0 SOMETHING" and then trying to access this drive.

I checked the code and found that it has been wrong since release 21.10.01. It always crashes when something else than REQUEST SENSE is sent because of an uninitialized pointer.
Looks as if those who once claimed in Discord that they were using SASI are no longer using it. Or they do not use current RaSCSI releases. IMO this shows that the time has come to get rid of this code.

akuker, 2022-03-06:
Its quite likely that users aren't updating with every version of RaSCSI. I'll bring it up in the discord dev channel and see what the reaction is.

akuker, 2022-03-06:
@uweseimet - Unless we get a strong reaction from the community - I propose that version 22.03 is the last to support SASI. If someone really needs SASI, they're probably not going to be using some of the new features that are in work. So, they could use GIMONS version, or just fall back to 22.03.

uweseimet, 2022-03-06:
@akuker OK, let's see whether there is any reaction. If somebody complains she/he should be willing to run regular tests. IMO there should be a reasonable number of users to justify keeping the SASI code. This code makes it harder to add SCSI improvements. But even better SCSI support would probably be more relevant and result in more potential RaSCSI users than keeping SASI support instead.

akuker, 2022-03-06:
I want to merge in the "caching" changes, then I think I'm ready to release. By default, there won't be a change in operation - the other method of caching will only be enabled if you add a command line parameter.

uweseimet, 2022-03-06:
Closing this ticket because with https://github.com/akuker/RASCSI/pull/724 the code has changed in a way that this ticket is no longer appropriate.

closed 2022-03-06 by uweseimet

uweseimet, 2022-03-06:
@akuker Any idea when the caching changes will be available? Looks as if there has not been any activity in the respective branch for thw last two weeks.
There are already a lot of user visible changes in the develop branch, i.e. a missing feature should probably not postpone the release for too long, if there is not enough time to work on this feature.