PiSCSI/piscsi #661: Akai S1000 stuttering audio when using raSCSI

ghal30n, 2022-02-10:
# Info

- Which version of Pi are you using:  Raspberry Pi Zero 2 W
- Which github revision of software:  21.12.01
- Which board version: V2
- Which computer is the RaSCSI connected to: Windows 10 x64 PC

# Describe the issue

When saving or loading any samples from raSCSI onto an Akai S1000, the audio stutters and clicks on playback.  Tried this with multiple emulated drives in different sizes.  

The raSCSI is confirmed working, I tried it on my S3000XL and it was working perfectly.

S1000 is confirmed working as well,  I installed a SCSI2SD V5.2 and it worked perfectly.

uweseimet, 2022-08-27:
@ghal30n You have not yet provided feedback. Please use the current release 22.07.02 or the develop branch for further testing.
Please note that we will have to discard this ticket if there is no further feedback.

uweseimet, 2022-09-10:
@ghal30n I would like to remind you to provde feedback, please see my previous comments for details. Otherwise I'm afraid we will have to close this ticket.

ghal30n, 2022-09-10:
> 

its all good! you can close the ticket, I'm way too busy to give this proper attention at the moment

uweseimet, 2022-09-10:
@ghal30n Thank you for your response. I will close the ticket. In case you have more time again (in the future) you might want to get back to this ticket.

closed 2022-09-10 by uweseimet