PiSCSI/piscsi #664: Host services extension: Printing service

uweseimet, 2022-02-12:
If somebody wants to print with a vintage computer, this might fail because the computer has not network and only a parallel printer port. Printers with parallel port are rare. (Are there any at all anymore?)
Software usually supports printing to a file instead of a printer, and in this case RaSCSI may be able to offer help.

One or two new SCSI commands for the host services (SCHS) device shall be implemented in order to support printing files sent by a client computer.
Suggested approach:

1. In a first step the client requests a job/file ID from the host services. This file ID is a string, which for the client is opaque. (It might just be a relative filename.)
2. The client sends the file data to RaSCSI. This may happen in several chunks. Each chunk contains the job ID, the total number of chunks and a running number for each chunk. Based on this information RaSCSI knows whether a file has completely been transferred. Chunks must be sent in order.
3. When the transfer is done the client sends another SCSI command in order to trigger printing. The job/file ID to print has to be passed with this command.

All uploaded files are saved into a special folder. These files are meant to be temporary, i.e. rascsi clears the folder on startup. When creating the SCHS device a parameter for the printer to be used can be passed, e.g. "lp=SOMETHING".

The SCSI commands used might be custom commands. WRITE LONG might also qualify because its sementics deal with a sequence of bytes.

The benefit of this two step process (first send/upload the file, then print it with a *separate* command) is the potential of doing other things with the uploaded file, i.e. not only printing.

uweseimet, 2022-02-12:
@akuker @rdmark I am interested in your feedback. Technically this should work fine. Are there any issues with the approach (job ID etc.)?

rdmark, 2022-02-13:
@uweseimet This is a neat and potentially very useful idea, in particular for platforms that don't have a network printer spooler. For Macintosh it's slightly less valuable, since it comes with Apple's LaserWriter drivers which can be used over Ethernet with a modern CUPS printer server via, say, Netatalk's papd daemon. It's relatively straight-forward to set up network printing: https://github.com/akuker/RASCSI/wiki/AFP-File-Sharing#Print_Server

I don't have first-hand experience with SCSI printers on Mac, but after some cursory research I ran into the MicroDry line of printers from ALPS. There are references to drivers for Mac (and Windows) but I haven't been able to find them archived yet:

https://aikishugyo.no-ip.org/alps/maint/mac-scsi.html
http://www.alps-printer.com/printers-md5000-c-3_12.html
https://web.archive.org/web/20170322022853/https://www.alps.co.jp/brand/printer/drive/main.htm

rdmark, 2022-02-13:
@uweseimet Have you considered creating an interface between the RaSCSI printer device and [CUPS](http://www.cups.org/documentation.html)? This would give you plug'n'play network printing to any CUPS/AirPrint compatible printer that resides on the same network, potentially.

uweseimet, 2022-02-13:
@rdmark An interface to CUPS might be a second step. Not sure, though, whether it would be worth the effort.

rdmark, 2022-02-13:
Two arguments for this feature:
- When I got papd working seamlessly in my fork of Netatalk, there was a surge of interest and people sharing stories about printing from their old Macs to modern printers. Again, this may simply be because of the novelty of it, but at the very least it could be another "marketing" talking point for RaSCSI to raise awareness and interest in the project. :)
- Arguably, the by far broadest potential utility for vintage computers in this day and age, beyond the hobbyist aspect, is word processing. Word processing and printing go hand in hand. I've seen articles and blog posts over the years about a vintage computer as being an ideal tool for an author, what with the lack of distractions, so the idea is not entirely novel.

uweseimet, 2022-02-13:
@rdmark Regarding workd processing you are absolutely right. There were popular word processing applications for the Atari, and usually you cannot just convert files in their proprietary format to, let's say, OOXml, in order to print it from a current platform.
Once more I have to point out the situation with the Atari, though: Besides excellent software emulators there are also some clones, i.e. you do not necessarily need an original Atari machine in order to be able to print with a word processing application for the Atari. Instead, you can run your Atari emulator, print to a file from within the emulator, and then use your native OS to print this file. This is not the real thing, of course ;-).

uweseimet, 2022-02-13:
There is at least one feature that is required but currently not support: Transfers of data chunks that are no multiples of 512. The printer device requires this feature. When it is present there is also a better chance to improve the currently incomplete implementation of READ/WRITE LONG, which also works on data chunks of any size.

uweseimet, 2022-02-15:
@rdmark I was faster than expected. You might want to add support for the SCLP device to the web UI. Two optional parameters are supported, which is reported by the protobuf interface, including their defaults.

```
  SCLP  Supported LUN numbers: 0-31
        Parameter support
        Default parameters: cmd=lp -oraw, timeout=30
```

akuker, 2022-02-17:
Great job, @uweseimet !!! This is a very cool feature! Thank you for adding it!

Do you have any documentation that we could put on the wiki for Atari users to take advantage of this new functionality? The documentation in the code is probably too low level for 99% of the users.

akuker, 2022-02-17:
Tonight I learned that Apple made a SCSI Laser Printer..... so looks like there's going to be a Phase 2 to this project. ;] Hopefully I can find one!!

https://lowendmac.com/1988/laserwriter-iisc/

uweseimet, 2022-02-17:
@akuker The RaSCSI printer implementation is not Atari-specific. I claim (but can be wrong), that it is a generic SCSI printer with a minimum set of features. I am sure that the Apple printer is more specific, and it should be possible to extend the existing implemenation accordingly, provided there is documentation on the Apple printer.
Regarding user documentation, this documentation will have to be part of the printer drivers, because each platform has its specific approach on how to print to such a printer. I assume that usually you just install SCSI printer drivers, just like you install your drivers for the Daynaport.
Regarding the Atari in particular I will publish documentation together with my set of RaSCSI client tools for the Atari (realtime clock, shutdown, printer). I intend to release this tools as soon an official RaSCSI release with the new features is available.

closed 2022-02-17 by uweseimet

uweseimet, 2022-02-20:
@akuker Do you know where to get a driver/system extension for a LaserWriter SC? I only can find extensions for printers with network support, but not with SCSI support. But such drivers must exist for the two SCSI printer models once offered by Apple.

marciot, 2022-04-09:
I managed to track down some documentation on implementing print drivers for classic Mac OS and added it to this page [SCSI Printer](https://github.com/akuker/RASCSI/wiki/SCSI-Printer)

marciot, 2022-04-09:
Does this SCLP driver take Postscript or bitmaps? Writing a Mac OS PostScript driver from scratch would be a herculean task, but it may be possible to use the documentation in the Learning to Drive document to implement a simple driver that sent over a B&W bitmap for printing, similar to the ImageWriter driver.

marciot, 2022-04-09:
Also, it looks like Apple actually made *two* SCSI printers. They made the [LaserWriter SC], which uses bitmaps like the ImageWriter while the [LaserWriter IISC] is a PostScript printer. The Learning to Drive document does mention that the low-level print drivers can either support text (i.e. postscript) or bitmaps.

[LaserWriter SC]: https://lowendmac.com/1990/personal-laserwriter-sc
[LaserWriter IISC]: https://lowendmac.com/1988/laserwriter-iisc

uweseimet, 2022-04-09:
@marciot It takes raw printer data. Please see scsi_printer.cpp for details.

marciot, 2022-04-10:
So I wrote the developer of PrintToPDF, a classic Mac OS print driver, and asked for the source code for that. He sent me his work, but not the source for a PDF library he did not own, so it is a start. With the documentation I found earlier, I might now have enough info to implement a Mac OS print driver from scratch. A couple of observations, however:

- Converting QuickDraw calls to vector-based PostScript would be a **huge** undertaking. I believe emulating the LaserWriter IISC and using Apple drivers would be the best solution for high-quality print graphics.
- Making a quick and dirty QuickDraw bitmap print driver seems like it might be within the scope of a one-man project, however.
- Since the SCLP driver doesn't know about bitmaps, perhaps one approach is to [embed a bitmap] inside a Postscript file. This does not appear to be hard.

[embed a bitmap]: http://paulbourke.net/dataformats/postscript/

akuker, 2022-04-13:
@marciot - PotatoFi has tracked down and LaserWriter II, and I found a LaserWriter IISC controller board on ebay, so I'm thinking of going down the LaswerWriter IISC route. I just need to figure out how to get a LaserWriter half-way across the US for a reasonable price :)