PiSCSI/piscsi #668: SGI Stray Interrupt

dillera, 2022-02-14:
# Info

- Which version of Pi are you using:
- Zero
- Which github revision of software:
- RaSCSI version: 21.10.1 [cf9a578]
- Which board version:
- full board
- Which computer is the RaSCSI connected to:
SGI Personal Iris 4D/20

# Describe the issue

SGI logs:

sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0

Happens with both a Personal Iris and Indigo scsi bus.

Debug logs on RASCSI:

```
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.468] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.468] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000979 blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.472] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.472] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.472] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000979 blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.476] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.476] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.476] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000979 blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.480] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.480] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.480] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000A1A blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.484] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.485] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.485] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000A1A blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.489] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.489] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.489] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000979 blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.493] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.493] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.493] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000979 blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.497] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.497] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.497] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000979 blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.501] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.501] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.501] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000A1B blocks=1
Feb 14 12:21:06 rascsi2 RASCSI[421]: [2022-02-14 12:20:47.505] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
```

dillera, 2022-02-14:
It happens pretty regularly, but things do progress, here is output from a SGI:

```
There is no valid file system on /dev/dsk/dks0d2s6

sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
Make new file system on /dev/dsk/dks0d2s6? yes
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
sc0: Unexpected info phase, state 4A phase 0
sc0: Resetting SCSI bus: stray interrupt error, phase 0x0, state 0x4A
...
```

uweseimet, 2022-08-27:
@dillera I just stumbled upon this ticket. Would be great if you could verify with the latest release 22.07.02 or the develop branch whether the problem is still there, and if it is please add a rascsi log on *trace* level.

uweseimet, 2022-09-10:
@dillera I would like to remind you to provde feedback, please see my previous comments for details. Otherwise I'm afraid we will have to close this ticket.

uweseimet, 2022-09-12:
Closed because without the required feedback working on this ticket is not possible.

closed 2022-09-12 by uweseimet