PiSCSI/piscsi #672: Electrical short between DB50 connector and PCB through-holes?

marciot, 2022-02-15:
# Info

- Which version of Pi are you using: N/A
- Which github revision of software: N/A
- Which board version: Self-assembled kit FULLSPEC 2.4a
- Which computer is the RaSCSI connected to: N/A

# Describe the issue

I had a RaSCSI that was working fine for many months, but recently the power light would turn off occasionally and the board would stop working. Flexing the board would cause it to come on again. I eventually figured that whatever was going on was happening on the DB50 side. I attempted to reflow all the solder joints and did this not help. Eventually, it occurred to me that there might be a short between the DB50 itself and the through-holes right underneath it. I shimmed a strip of paper between the DB50 and the PCB and the problem has gone away.

I am reporting this to ask whether anyone else has seen this issue. I don't know if all boards have those through-holes under the DB50, as the one pictured in the assembly instruction do not. I also do not know if there is anything electrically conductive underneath that DB50. I seem to recall not, but considering the piece of paper solved the problem I have no other explanation.

marciot, 2022-02-17:
@akuker: Yup, that’s what I meant (sorry, didn’t count the pins). Anyhow, it seems like the underside of the connector is plastic except at the metal end, so I’m not exactly sure why this is causing a problem, especially since it appears everything on the PCB at that end is ground. It could be something in particular about my build, for all I know. But inserting a piece of paper there did solve the issue for me so I thought I would mention it.

marciot, 2022-02-17:
@akuker: Well, I suppose the PCB appears to be ground on the bottom side that is visible. What is on the other side of the PCB? Is there any chance there is a 5V running to the edge of the PCB, nearest the metal edge of the connector? This only started to happen for me after a few months of use, so I wonder if the metal edge scraped away the solder mask on the top side of the board and started to make contact with whatever was underneath.

akuker, 2022-02-17:
You called it!! The 5v line runs right next to that metal connector. 
[image]

Could you try cutting this trace to see if that solves the issue? As long as you're not using the Powerbook connector, there won't be any negative effect.
[image]

rdmark, 2022-08-19:
Closing since a DIY fix has been identified, and future hardware designs improved to avoid the fail state.

closed 2022-08-19 by rdmark

jfredrickson, 2022-09-21:
Just to clarify, does 2.5b fix this?

akuker, 2022-09-22:
I don’t believe so. Unfortunately, we found this issue after the 2.5 boards were already ordered. The 2.6 version will be available next week, which does have the fix for this.

jfredrickson, 2022-09-23:
Is it possible that this bug could have somehow wrecked the SCSI controller on my SPARCstation 10 when the 5V shorted to ground?

I'd been using the RaSCSI successfully for a few months on my Mac and SGI. I recently hooked it up to the Sun and it was working well at first. But due to the physical position of the computer and the RaSCSI, there was more downward pressure on the DB25 connector this time, so I started to get SCSI errors, which led me to discover this issue. Now the Sun cannot access the RaSCSI or even the internal SCSI hard disk. It can see the devices, but I get bus parity errors anytime I try to do something. So I'm worried this caused permanent damage. :(