PiSCSI/piscsi #685: Remove NetBSD support?

uweseimet, 2022-02-19:
Is NetBSD support still required, and is anybody testing this? If not it should be removed. This would most likely also remote the need for linking with zlib, because only the ctapdriver uses it for the crc32 checksum, and Linux has its own implementation for calculating the crc32 checksum.

uweseimet, 2022-02-19:
@akuker As usual, when I propose to remove something, I would like to get your feedback ;-).

akuker, 2022-02-22:
The NetBSD support was added for building on MacOS. I added this when I was developing a feature a long time back. I don't think its necessary. Feel free to remove it if you want.

rdmark, 2022-02-22:
FWIW, I was speaking to someone the other day who said they were working on making the RaSCSI run properly on NetBSD.

uweseimet, 2022-02-22:
@rdmark If they do this I am quite sure that would have to start from scratch. It's very unlikely that the current fragments of NetBSD support are still useful or even working after the huge changes since last summer, and the fragments might even be MacOS-specific. Somebody who would want to support NetBSD would have to verify any changes, by the way, because she/he cannot expect that we test every change on NetBSD.
@akuker I am going to remove the existing #ifdefs and the related code. It's not that much as far as I can tell.

closed 2022-02-23 by uweseimet