PiSCSI/piscsi #691: Releasing reserved IDs in rasctl is not user friendly

rdmark, 2022-02-21:
In the latest develop (and earlier):

From reading the man page for rasctl for the '-r' option, it is not clear how to release reservations. Through experimentation, ```rasctl -r ""``` seems to release all reservations. I think either updating the documentation, or introducing a separate  "release reservations" command would be appropriate.

rdmark, 2022-02-21:
@uweseimet What are your thoughts here?

uweseimet, 2022-02-21:
@rdmark I just updated the manpages which a note on the meaning of an empty list. I would not want to add a new command for this, because with the existing command you can easily release all reservations. If you know how :). But that's now mentioned in the manpage.

uweseimet, 2022-02-21:
@rdmark One more note on this:

rdmark, 2022-02-22:
@uweseimet Thanks for the quick fix!

closed 2022-02-22 by rdmark