PiSCSI/piscsi #705: Can't compile Netatalk

Deftaudio, 2022-02-25:
# Info

- Which version of Pi are you using:
Pi3b
- Which github revision of software:
v22.02.01
- Which board version:
FULL
- Which computer is the RaSCSI connected to:
N/A

# Describe the issue
I'm having problems in compiling Netatalk, It finished with the error 
```
Making all in initscripts
make[3]: Entering directory '/home/pi/Netatalk-2.x/distrib/initscripts'
make[3]: *** No rule to make target 'a2boot.service', needed by 'all-am'.  Stop.
make[3]: Leaving directory '/home/pi/Netatalk-2.x/distrib/initscripts'
make[2]: *** [Makefile:426: all-recursive] Error 1
make[2]: Leaving directory '/home/pi/Netatalk-2.x/distrib'
make[1]: *** [Makefile:479: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/Netatalk-2.x'
make: *** [Makefile:409: all] Error 2
```
I also tried not to enable a2boot, and just configure it with 
`./configure --enable-systemd --enable-zeroconf`
But that didn't fix the problem. Please, suggest what did I miss.

Deftaudio, 2022-02-27:
It has been addressed here: 
https://github.com/rdmark/Netatalk-2.x/issues/6

Documentation needs to be updated though. New service names don't include `netatalk` prefix. So atalkd, papd and etc...

rdmark, 2022-02-27:
@Deftaudio Good catch re: the wiki documentation. I've updated it.

closed 2022-02-27 by rdmark