PiSCSI/piscsi #723: SASI FORMAT command probably has wrong opcode

uweseimet, 2022-03-05:
According to http://www.bitsavers.org/pdf/ansi/X3T9/X3T9.3_185_RevE_SASI_Apr82.pdf the FORMAT command has opcode 4, just like with SCSI.

uweseimet, 2022-03-05:
@akuker I would like to change the current opcode, because the SASI specification says it is 0x04.

Believe it or not, I just realized by looking at the specs that I have a SASI drive at home. Some old drives distributed by Atari for the Atari ST series are MFM and RLL drives with an Adaptec ACB-4000 controller, which appears to be a SASI-to-MFM/RLL bridge. So it's not surprising that this drive appears to support the same set of commands RaSCSI supports for SASI.
My drive is still fully functional and it looks like new, because I have hardly used it. But my Atari driver software supports it, of course, and this is why I know for sure that for this drive the opcode for FORMAT is 0x04, just like the SASI specification says ;-). Not 0x06, as currently used by RaSCSI.

akuker, 2022-03-06:
Good catch Uwe! I made a comment in the pull request that the original RaSCSI code treated both commands (0x04 and 0x06) as the format command.

closed 2022-03-06 by uweseimet