PiSCSI/piscsi #729: Remove legacy SASI support after 22.03 release

uweseimet, 2022-03-06:
As discussed in https://github.com/akuker/RASCSI/issues/660 there does not appear to be much interest in SASI support. Serious bugs (segfaults) have not been noticed for 5 months, i.e. nobody used SASI drives with the RaSCSI versions released in the last 5 months. Or at least nobody had any interest in reporting these bugs, or in testing the SASI emulation.
In addition, the fact that the SCSI controller is a subclass of the SASI controller makes it hard to extend the SCSI code in order to address existing tickets, or to add new functionality. For SASI, on the other hand, not no features can be expected. There are a lot of other issues with the SASI code that makes it a maintenance burden, as one can see from existing and old tickets.

This is why after the 22.03 release SASI support should be removed.

philpem, 2022-03-15:
I have a possible use case for SASI support -- the Acorn Domesday system uses it to communicate with the Laserdisc drive.
My aim is to have RaSCSI pick off the Laserdisc commands and send them to a framestore (or more likely a full-fat Pi) to play the audio and video streams and still-frames.

My use case could be accommodated with a branch or fork if needs be -- it is currently a speculative development, but I thought it worth a mention as others may have their use cases.

uweseimet, 2022-03-15:
@philpem Hi. Can you please add a link to a documentation (with the SASI commands it uses) of this drive? Any suggestions how to decouple the current SASI code (in particular the controller) from the SCSI code?
As mentioned in my initial comment the SASI code has been broken for about half a year. Maybe you can indicate why you have not updated (at least that is what I assume) for such a long time?

philpem, 2022-03-15:
The VP415 operating manual is here: https://www.domesday86.com/wp-content/uploads/2017/01/Philips-VP415-Operating-Instructions.pdf -- the "F-code" commands are used to program the player, and the following chapter explains how to send them over SCSI.

The Domesday86 project has more information on the player, including:

Other VP415 documentation and internals: https://www.domesday86.com/?page_id=316
The BeebSCSI controller, including an implementation of the VP415 commands: https://www.domesday86.com/?page_id=689

It appears that I was mistaken about these being SASI drives -- I must have been thinking about the earlier Acorn Winchester Host Adapter, which uses an Adaptec or Xebec SASI-to-MFM bridge controller.

I have only recently discovered this project, so you are sadly mistaken -- I have never worked on the SASI code. In any case, it seems like I need SCSI, not SASI.

If anyone wanted to work on SASI, there is a description of it in the CDC Wren II Interface Specification: http://bitsavers.trailing-edge.com/pdf/cdc/discs/wren/77738148B_Interface_Specification_for_WREN_II-SCSI_SASI_SUBSET_94151_198511.pdf
But it seems from my limited research that it would mostly be useful for the Japanese X68000 machines.

drdpj, 2022-03-21:
Just to note, there's going to be a bit of concerted effort about to take place to make this work with a victor 9000 which used a Xebec S1410 - so please don't lose the SASI stuff!

uweseimet, 2022-03-22:
@drdpj Any idea why nobody who is interested in SASI has used any RaSCSI release for more than half a year? Since that time SASI was definitely broken. Maybe you can indicate why you have not used any RaSCSI version since 21.10?

drdpj, 2022-03-22:
Because people have only just started getting these machines working (up
until very recently it was very hard to make floppy images for them), and
now we can get them booted it's an opportunity to look at mass storage.
Scsi2sd isn't an open system, but this is. Retro communities take time to
start adopting bits of hardware - removing functionality would seem to be
something of a retrograde step for these things, usually expanding
usefulness to futher platforms is the route they take (eg, flashfloppy).
Personally speaking, I'm waiting for my rascsi adapters to materialise from
the far east and will start to poke at it properly when those arrive.
Fozztexx has already started. Also worth noting that raspberry pis are hard
to source at a sensible price at the moment, so that delays adoption.

uweseimet, 2022-03-22:
@drdpj SCSI2SD is pure SCSI, not SASI, isn't it? So it would not qualify for SASI anyway. Please correct me if I am wrong. Which SASI hardware are you using? You can probably help with testing RaSCSI wrt to SASI? Up to now nobody else has been interested in that.
Would you be willing to work on the SASI code, in order to decouple it from the SCSI code? The problem with the SASI code is that it interferes with the further development of the SCSI code.

drdpj, 2022-03-22:
@uweseimet SCSI2SD does do quite a bit of SASI, and works where some very early SASI/SCSIesque drives were used (I use one in an SJ Research MDFS, and they've also been used in Acorn Filestores) - big list here: https://www.codesrc.com/mediawiki/index.php/SCSI2SD_V5_Compatibility

I was working with Michael M to try and get it to emulate the Xebec controller properly (that's the PDF I linked) so it would work in the Victor 9000 (that's my particular use case for this, and it's very picky), but as it's a closed project the testing was a very to-and-fro thing taking a long time and eventually it reached a dead end. I would very happily help test the SASI code on my system, and also to develop it further. Decoupling I might not be so useful (I've got Java experience, not so much C++), but I can certainly help think about how best to decouple it although actually following through on that myself might be trickier!  It seems that both SCSI and SASI should be subclasses of "controller" rather than SCSI being a subclass of SASI.

uweseimet, 2022-03-22:
@drdpj Do you have time to test the SASI code of the upcoming 22.03 release, i.e. the current develop branch? Looks as if the release is stalling, so there may still be enough time for testing. In general, somebody regularly testing the SASI develop code would be important.

drdpj, 2022-03-22:
@uweseimet yes - although I'm not sure when my adapters will turn up! The buffers are currently stuck in customs and the cards are somewhere in China. I also strongly suspect that the Xebec is a special case of SASI. However, if it behaves on the victor - which I know is quite picky - it should behave with other things that are using that same Xebec controller.

uweseimet, 2022-03-22:
@drdpj Well, if you have a special case of SASI it might not work with RaSCSI without any changes to the existing code. Looks as if you may have no choice but to get more familiar with C++ ;-). All in all, somebody would have to maintain the code.

drdpj, 2022-03-22:
:D - I expect it may well need some changes, so at some stage I'm going to have to get my hands dirty with C++! Thing about SASI is it's not quite a uniform standard so far as I can tell, and there were a number of slightly different implementations. Things very much firmed up with SCSI.

uweseimet, 2022-04-26:
@drdpj Any news? You are still the only one who has expressed interest in SASI support.

drdpj, 2022-04-26:
There is a fork that is working that fozztexx has done but I don't know if
it's resulted in a pull request?

d.

uweseimet, 2022-04-26:
@drdpj I'm afraid I don't know which fork you are talking about. Why should there be a pull request? What you you mean with "fork that is working"? Are there any issues with the current RaSCSI development branch?

drdpj, 2022-04-26:
I would have hoped there'd be a pull request to bring the fixes back into
the main source. This fork: https://github.com/FozzTexx/RASCSI/tree/Victor9k
has fixed issues with SASI support in the main branch and got it working
with machines that expect the Xebec SASI controller.

uweseimet, 2022-04-26:
@drdpj There is no related PR ...

drdpj, 2022-04-26:
I can create one? I've just merged the changes in with the master branch on
my own fork?  Which branch should I request a merge with?
d.

uweseimet, 2022-04-26:
@drdpj Please first create a ticket that describes the bug/issue your changes are going to fix. Your PR has to be raised against the develop branch, not the master branch. So before creating the PR please ensure that your changes refer to the develop branch, and that the changes have successfully been tested. Note that the SASI code also may affect the SCSI code, i.e. testing with SCSI devices may also be required.

drdpj, 2022-04-26:
They're not my changes so I'll do my best to describe. I think a memory
leak was fixed with the SASI emulation too - but I couldn't say where.

uweseimet, 2022-04-26:
@drdpj Well, if you raise a PR you would have to know the details and you need to be able to address any issues with the PR. If you are not familiar with the changes IMHO a PR does not make much sense.

drdpj, 2022-04-26:
Agreed, so I shall try and get the person who made the changes to issue one.

uweseimet, 2022-04-26:
@drdpj Yes, this would be be. We need a proper ticket and PR against the develop branch, and the changes must have been tested against the develop branch. And it would be very good to test first whether any changes are needed at all, because the develop branch already contains numerous bugfixes related to SASI.

rdmark, 2022-05-11:
@uweseimet A stable release has been tagged and released now, so deprecating SASI should be considered for the roadmap of the next stable release I think!

uweseimet, 2022-08-22:
@rdmark The changes for this ticket may also affect the web interface. You can verify this in the sasi_removal branch, where the SASI device type has been removed from the protobuf interface. I recommend that you check whether your Python code relies on this type to be present. In general, clients should be able to cope with a type not being present anymore, i.e. except for being re-compiled against the modified protobuf interface no further action should be necessary. In particular, if a client manages configuration data, the client should not break when loading invalid configurations, e.g. configurations still containing unsupported device types.

uweseimet, 2022-08-28:
@Pacjunk Do you have time to test the latest sasi_removal branch? I have already successfully tested it and would like it to be merged into develop right after the 22.08 release. I have already prepared the changed required for .hd1 (SCSI-1) image file support, which are only a minor change on top of the sasi_removal branch.

Pacjunk, 2022-08-29:
@uweseimet Works fine on my system.

uweseimet, 2022-08-29:
@Pacjunk Thank you for your feedback. As soon as @akuker or @rdmark have approved https://github.com/akuker/RASCSI/pull/806 we can continue with the SCSI level changes.

uweseimet, 2022-09-03:
Summary the most important changes:

- Removed the SASI controller code
- New controller management. There is a new controller base class AbstractController and a class ControllerManager managing the controller lifecycle. The lifecycle management was removed from rasci.cpp and is covered by unit tests.
- New device management. The DeviceFactory manages the device lifecycle instead of rascsi.cpp. The new code is covered by unit tests.
- The lifecycle managment uses C++ collections with variable size instead of arrays with hard-coded sizes.
- The ScsiController method contains most of what was previously contained in scsidev_ctrl.cpp plus the code from sasidev_ctrl.cpp that was relevant for SCSI.
- scsi_command_util contains helper methods used for identical SCSI command implementations of more than one device
- Devices know their controllers, so that the controller instance does not need to be passed to each SCSI command. This change helps to decouple the devices from the controller. The phase_handler interface is also part of this decoupling.
- Use scsi_command_exception for propagating SCSI command execution errors, This resolves issues with the previous error handling, which was based on return values and often on magic numbers.
- Removed legacy SCSI error codes, all errors are now encoded by sense_key::, asc:: and status::.
- Fixed various warnings reported with -Wextra, -Weffc++ and -Wpedantic.
- Use constructor member initialization lists (recommended for ISO C++)
- Consistently use new/delete instead of malloc/free (recommended for ISO C++), resulting in better type safety and error handling
- Replaced variable sized arrays on the stack (violates ISO C++ and can cause a stack overflow)
- Replaced NULL by nullptr (recommended for C++), resulting in better type safety
- Use more const member functions in order to avoid side effects
- The format device page can now also be changed for hard disk drives (Fujitsu M2624S supports this, for instance), not just for MOs.
- Better encapsulation, updated access specifiers in many places
- Removed unused methods and method arguments
- Fixed a number of TODOs
- Added/updated unit tests for a lot of non-legacy classes
- Makefile support for creating HTML coverage reports with lcov/genhtml

closed 2022-09-03 by uweseimet