PiSCSI/piscsi #736: Issues with Akai MPC60 sampler

fatbit1, 2022-03-14:
# Info

- Which version of Pi are you using: PI0W Linux raspberrypi 5.10.17+
- Which github revision of software: version 22.03 --DEVELOPMENT BUILD-- (Mar  8 2022, 14:28:25) 22.3.-1 [9099d72]
- Which board version: FULL SPEC
- Which computer is the RaSCSI connected to: AKAI MPC60 Sampler (MARION SCSI BOARD) 

Thank you for looking into this. I found the following issues while  trying RASCI with the MPC60:
- different block sizes. Although 512 seems to provide the better results, there are still issues.'
- QUANTUM image type or other types of images
- Different cables, including a direct connection with the internal 25 to 50 pin ribbon cable.
- Formatting seems to work
- Saving a File: Sampler shows a blank screen
- Opening an existing image: Sometimes I get SW 06h on sampler screen , some other times it works, sometimes it gets stuck with no error
- Reading files. I believe I have been able to read very small files ( i.e. default parameter file) but it gets stucks with all the other files.
[rascsi.log](https://github.com/akuker/RASCSI/files/8248616/rascsi.log)

not-a-numb3r, 2022-03-20:
I have an MPC60 II as well. Please let me know if I can help in some way.

uweseimet, 2022-08-27:
@fatbit1 Can you please attach a smaller logfile, which is just capturing a particular, reproducable issue? Please use the latest release 22.07.02 or the develop branch for any further testing.

uweseimet, 2022-09-10:
@fatbit1 I would like to remind you to provde feedback, please see my previous comments for details. Otherwise I'm afraid we will have to close this ticket.

uweseimet, 2022-09-12:
Closed because without the required feedback working on this ticket is not possible.

closed 2022-09-12 by uweseimet

fatbit1, 2022-09-12:
Hi Uwe

Sorry for the late reply. Unfortunately, I will not have access to the MPC60 for 2 weeks.
I will update the thread as soon as I will be able to collect the requested files.
Thanks!

fatbit1, 2022-12-26:
Sorry for my later reply, Here is the piscsi.log file after trying to access 2 different existing images and formatting the third one.
Thanks

[piscsi.log](https://github.com/PiSCSI/piscsi/files/10304788/piscsi.log)