PiSCSI/piscsi #745: Out of memory compiling RaSCSI on Bullseye on Raspberry Pi Zero W

marciot, 2022-04-09:
# Info

- Which version of Pi are you using: Raspberry Pi Zero W
- Which github revision of software: develop
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: Macintosh Plus or LC II

# Describe the issue

My 8GB SD card had barely enough memory for disk images, so I decided to migrate to a larger card and start with a fresh install of RaspberryOS.

I just wanted to report that compilation of RaSCSI is failing on my Raspberry Pi Zero W, when using Bullseye, due to insufficient memory. I even tried disabling the desktop environment. I might try redoing this with Buster, which is what I was using before.

I don't imagine there is anything you guys can do about this, but I thought I would report it for just so it is logged somewhere.

I definitely would like to get my hands on a Raspberry Pi Zero W 2, as the older one seems barely adequate for RaSCSI at this point ☹️

uweseimet, 2022-04-09:
@marciot If this is due to insufficient RAM (this is how I read your ticket) you can use a larger swap partition or add a swap file.
I am going to close this ticket because just as you said one cannot do anything about it except for providing more memory, which in the case of a Pi means to increase the available swap space.

closed 2022-04-09 by uweseimet

rdmark, 2022-04-09:
@marciot I compiled RaSCSI on my Zero W most recently maybe a month ago, develop branch (which hasn't changed much since) and Bullseye Lite. Have you tried the Lite OS image?

marciot, 2022-04-09:
@rdmark: I'll try Bullseye Lite and see if I have better luck with it.

marciot, 2022-04-10:
@rdmark: It looks like using Bullseye Lite did the trick. Apparently using the full desktop distro, and then disabling the UI is not sufficient. That's a bit peculiar. The other thing that might matter is that the first time around I didn't know about the option in "raspi-config" to grow the volume to fit the SD card. Maybe that also increases swap space. I wonder if it would have worked if I had run that on the full desktop install.

I wonder why compiling RaSCSI is taking so much memory. Anecdotally I've heard that at one point the Linux maintainers reorganized their header files and managed to reduce compilation overhead by a **huge** margin. Would it be worth the effort, maybe, maybe not?

marciot, 2022-04-10:
Just an FYI, when configuring the web interface, I see this in the `sudo journalctl -f`.

```
Apr 10 19:18:08 raspberrypi RASCSIWEB[3780]: Starting web server for RaSCSI Web Interface...
Apr 10 19:18:18 raspberrypi RASCSIWEB[3852]: WARNING:root:The default locale could not be detected. Falling back to English.
Apr 10 19:18:18 raspberrypi kernel: bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
```

I'm not sure if this indicates some work will need to be done to the bridge code to support Bullseye. I generally use SSH tunneling rather than bridging my networks so I probably won't actually test it.