PiSCSI/piscsi #748: Feature Request: Add "Miniscribe M8425" to "Create Empty Disk Image File"

marciot, 2022-04-10:
# Info

- Which version of Pi are you using: Raspberry Pi W
- Which github revision of software: develop
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: Mac Plus

# Describe the issue

Currently, there is no way to create an arbitrarily sized disk image that is recognized by "Apple HD SC Setup". I have found a workaround that relies on using the identifier for the "Miniscribe M8425" drive, but using a larger file than the original drive. This workaround is described in #747.

I suggest adding a "SCSI Hard Disk image (Miniscribe M8425) .hdm" to the pull-down menu and the following help text in the bulleted list:

* Miniscribe M8425 (.hdm) works with an unpatched Apple HD SC Setup, however for sizes other than 20MB it is necessary to click "Partition" and "Custom" to adjust the default partitioning map.

marciot, 2022-04-10:
Alternatively, the UI could be reworked to allow any drive to mimic the vendor information of the Miniscribe M8425. I know the functionality already exists by creating properties files.

rdmark, 2022-08-26:
@marciot How about this for a counter proposal: Add an input field to the bespoke drive image creation page that allows you to customize the image file size for any bespoke drive.

Code change in https://github.com/akuker/RASCSI/pull/813

uweseimet, 2022-08-26:
Please note that adding new file extensions for features that are already configurable is not how it should be. New drive types are reserved for new SCSI mass storage device types (I doubt that there can be any) or for something that cannot be made configurable in the rascsi core with standard means. Drive names, sector sizes etc. are already configurable (or can be made configurable) in the available clients, so please no new device types/file extensions for this.

rdmark, 2022-08-28:
My proposed solution is now in develop.

closed 2022-08-28 by rdmark