PiSCSI/piscsi #750: Not longer able to attach DynaPort device without network bridge.

marciot, 2022-04-17:
# Info

- Which version of Pi are you using: RaspBerry Pi W
- Which github revision of software: develop
- Which board version: FullSpec
- Which computer is the RaSCSI connected to:

# Describe the issue

With the older versions of RaSCSI, it was possible to start the DynaPort device without configuring a network bridge. It was then possible to secure shell into the RaSCSI and use port forwarding to connect to services running on the Macintosh via the "private" IP address. I used it this way because I had my RaSCSI configured in a DMZ, also acting as a web server, and I did not want to expose the Mac to the network at large.

With the latest version in "develop", it brings up a message "Configure IPv4 forwarding before using a wireless network device. Please follow the instructions at https://github.com/akuker/RASCSI/wiki/Dayna-Port-SCSI-Link" and then refuses to start up the DynaPort device.

I feel like this is a regression because it is assuming the user wants to do things a certain way. This message should be a suggestion, but not force the user to do a certain configuration.

pendleton115, 2022-04-17:
I am having the same problem on my hardware, Macintosh Classic, I've tried System 7.0.1, and 7.5.3. The Dayna Port hardware is not detected in the installer but shows up in SCSI Probe and the DaynaPort Hardware Diagnostic.

rdmark, 2022-04-17:
@marciot Please separate each of your issues into separate tickets! This is the only way to cleanly track and troubleshoot each of them.

With regards to the third issue, this is something that we introduced in the stable November release last year. See https://github.com/akuker/RASCSI/issues/407 . The reason that we're doing it like this is that if you don't have the bridge configured, attaching a network device may disrupt the Pi's network connectivity. A user who's not familiar with Linux networking may end up with a non-functional environment at that point. You may still attach it via rasctl if you know what you're doing.

The real solution to this would be to change the logic in the rascsi backend to somehow be aware of the network environment and act accordingly.

marciot, 2022-04-17:
I am going through the process of testing individual releases to see whether the DaynaPort drivers and certain prior disk images break. It takes a few hours to compile each release, but I will update this table as I go along:

|                                 | Disk Images | DaynaPort |
|----------------------|---------------|-------------|
| develop   (Bullseye) | Not all          | Fails          |
| v21.11.01 (Buster) | Not all             | OK            |
| v21.10.01 (Buster) | OK                  | OK             |

marciot, 2022-04-19:
Just an update on this thread. After trying a bunch of combinations, I think the only firm conclusion I came to is that the DaynaPort is not working on the develop branch for whatever reason. This is all. It works on all the *release* versions running on Buster *and* Bullseye, however.

I do not think there was anything in fact wrong with disk images. That was a red herring.  There are a bazillion factors that go into whether or not a particular disk image will work on a given Mac and I must have gotten confused about which ones worked on the Mac Plus to start with. The images which were supposedly had stopped working after my upgrade also did not work with my old RaSCSI build, so you can disregard my suggestion that disk images had begun failing after my upgrade. This was false. I now think they never worked in the first place.

marciot, 2022-04-19:
> I am having the same problem on my hardware, Macintosh Classic, I've tried System 7.0.1, and 7.5.3. The Dayna Port hardware is not detected in the installer but shows up in SCSI Probe and the DaynaPort Hardware Diagnostic.

@pendleton115: I've confirmed that the "develop" branch is not working, but the tagged releases seem to work fine. Can you verify what checkout of RaSCSI you are using?

rdmark, 2022-04-19:
I had a quick check with the latest develop on my Classic II / System 7.5.3 and Open Transport 1.1.4, using the wifi/NAT bridge. It works well in my testing so far. Will test on my SE tomorrow to check if it's a 68000 problem.

pendleton115, 2022-04-19:
I will have to check which version I am running, but I updated it recently using the "Easyinstall.sh" script. I would assume this would update to the most recent release build.

> @pendleton115: I've confirmed that the "develop" branch is not working, but the tagged releases seem to work fine. Can you verify what checkout of RaSCSI you are using?

pendleton115, 2022-04-19:
Just checked, the version I am using is RaSCSI version: 22.3.-1 [f2aa2db](https://github.com/akuker/RASCSI/commit/f2aa2db73f058153ce853d18d1946dc1b89cf510)

rdmark, 2022-04-21:
Ignore my previous comment. I could confirm that the DaynaPort device isn't able to initialize on neither my Classic II nor my SE. I The Classic II runs System 7.5.3 and Open Transport 1.1.4. The SE runs System 7.0.1 and MacTCP 2.1.

rdmark, 2022-04-21:
This is a trace log from when I attempt to initialize the network interface on my SE.

```
Apr 21 05:51:28 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.922] [trace] void SASIDEV::DataIn() Going into Data-in Phase
Apr 21 05:51:28 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.922] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 18
Apr 21 05:51:28 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.922] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
Apr 21 05:51:28 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.922] [trace] void SASIDEV::Status() Status phase
Apr 21 05:51:28 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.922] [trace] void SASIDEV::Status() Status Phase $00
Apr 21 05:51:28 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.922] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Apr 21 05:51:29 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.923] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Apr 21 05:51:29 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.923] [trace] void SASIDEV::MsgIn() Starting Message in phase
Apr 21 05:51:29 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.923] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Apr 21 05:51:29 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.923] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Apr 21 05:51:29 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.923] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Apr 21 05:51:29 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.923] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Apr 21 05:51:29 rascsiZW RASCSI[361]: [2022-04-21 05:51:28.923] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.915] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=3 (with device)
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.916] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.916] [trace] virtual void SASIDEV::Command() Command Phase
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.916] [trace] virtual void SASIDEV::Command() CDB[0]=$00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.916] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.916] [trace] virtual void SASIDEV::Command() CDB[2]=$00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.916] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.916] [trace] virtual void SASIDEV::Command() CDB[4]=$00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] virtual void SCSIDEV::Execute() Execution phase command $00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing TestUnitReady ($00)
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [debug] Error status: Sense Key $02, ASC $3A, ASCQ $00
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] bool PrimaryDevice::CheckReady() Device not ready
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] virtual void SCSIDEV::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] void SASIDEV::Status() Status phase
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] void SASIDEV::Status() Status Phase $02
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] void SASIDEV::MsgIn() Starting Message in phase
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.917] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.918] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.918] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.918] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.919] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=3 (with device)
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.919] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.919] [trace] virtual void SASIDEV::Command() Command Phase
Apr 21 05:51:30 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.920] [trace] virtual void SASIDEV::Command() CDB[0]=$03
[... cut ...]
Apr 21 05:51:31 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.921] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Apr 21 05:51:31 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.921] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Apr 21 05:51:31 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.921] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Apr 21 05:51:31 rascsiZW RASCSI[361]: [2022-04-21 05:51:30.921] [trace] virtual void SCSIDEV::BusFree() Bus free phase
```

rdmark, 2022-04-21:
@uweseimet So it seems we have a regression bug in develop. I think this should be addressed before we cut a release. I'll keep trying to narrow down when the regression happened using binary search later this week (work is keeping me busy....)

uweseimet, 2022-04-21:
@rdmark @marciot Yes, it would be good to know the commit that causes the problem.

rdmark, 2022-04-25:
@uweseimet Through binary search I narrowed it down to this commit to develop: 1df7cdb1f3cc70fce9b163db2dad5eae8dc87520

uweseimet, 2022-04-25:
@rdmark Is there any difference in the trace logs between these two versions?

@marciot There are not many changes related to the DaynaPort in this commit, mainly regarding INQUIRY. Since you know C++ you can probably find out what's wrong by reverting the DaynaPort changes step by step?

rdmark, 2022-04-25:
An example trace log from the same stage when it initializes the network interface on the Mac.

```
[2022-04-25 09:12:43.448] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-25 09:12:43.448] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-25 09:12:43.448] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-25 09:12:43.448] [trace] virtual void SASIDEV::Command() CDB[0]=$12
[2022-04-25 09:12:43.448] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-25 09:12:43.448] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-25 09:12:43.448] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2022-04-25 09:12:43.448] [trace] virtual void SASIDEV::Command() CDB[4]=$25
[2022-04-25 09:12:43.448] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2022-04-25 09:12:43.448] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2022-04-25 09:12:43.449] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2022-04-25 09:12:43.449] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing Inquiry ($12)
[2022-04-25 09:12:43.449] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2022-04-25 09:12:43.449] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 37
[2022-04-25 09:12:43.449] [trace] virtual int GPIOBUS::SendHandShake(BYTE*, int, int) DELAYING for 100us after 6 bytes
[2022-04-25 09:12:43.449] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2022-04-25 09:12:43.450] [trace] void SASIDEV::Status() Status phase
[2022-04-25 09:12:43.450] [trace] void SASIDEV::Status() Status Phase $00
[2022-04-25 09:12:43.450] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 09:12:43.450] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2022-04-25 09:12:43.450] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2022-04-25 09:12:43.450] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 09:12:43.450] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 09:12:43.450] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 09:12:43.450] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2022-04-25 09:12:43.450] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2022-04-25 09:12:43.451] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-25 09:12:43.451] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-25 09:12:43.451] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-25 09:12:43.451] [trace] virtual void SASIDEV::Command() CDB[0]=$0E
[2022-04-25 09:12:43.451] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-25 09:12:43.451] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-25 09:12:43.451] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2022-04-25 09:12:43.451] [trace] virtual void SASIDEV::Command() CDB[4]=$00
[2022-04-25 09:12:43.451] [trace] virtual void SASIDEV::Command() CDB[5]=$80
[2022-04-25 09:12:43.452] [trace] virtual void SCSIDEV::Execute() Execution phase command $0E
[2022-04-25 09:12:43.452] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $0E
[2022-04-25 09:12:43.452] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = SCSIDaynaPort; U = SASIDEV] Executing EnableInterface ($0E)
[2022-04-25 09:12:43.452] [debug] bool CTapDriver::Enable(): ip link set ras0 up
[2022-04-25 09:12:43.452] [info] The DaynaPort interface has been ENABLED.
```

rdmark, 2022-04-25:
@uweseimet The differences that I see between a failed and successful process is that different commands are being executed by Dispatcher. The failed case executes TestUnitReady and RequestSense which I couldn't find anywhere in the successful case. The successful case executes Inquiry and EnableInterface.

This could be that I'm looking at different spots in the logs, but I tried to capture them right around where it tries to initialize the interface, so they should be equivalent.

I've been capturing longer logs but only posting the parts that seem relevant.

marciot, 2022-04-25:
> @marciot There are not many changes related to the DaynaPort in this commit, mainly regarding INQUIRY. Since you can reproduce the problem and know C++ you can probably find out what's wrong by reverting the DaynaPort changes step by step?

I won't have the time to work on this until next weekend, and even then I am hobbled by the fact that it takes several hours to compile RaSCSI on my Raspberry Pi Zero W, due to the swap issue I mentioned earlier. If anyone wants to donate a Raspberry Pi 4 to the cause it would definitely help (not that there are any in stock anywhere right now).

marciot, 2022-04-25:
I would think a sensible approach for this bug is to revert this entire commit and then separate out the INQUIRY changes from all the other code refactoring changes.

uweseimet, 2022-04-25:
@marciot We cannot revert, because other commits are based on it, and reverting it causes numerous conflicts.

marciot, 2022-04-25:
Well, I'll see what I can do this weekend. But like I said, any preliminary work that can be done in advance on a more capable Raspberry Pi would be very helpful.

uweseimet, 2022-04-25:
@rdmark I will have a look at the logs, maybe I can find something. Currently I don't have a working network setup, so that I can't test myself. Correct me if I am wrong:

uweseimet, 2022-04-25:
@rdmark Looks as if your log contains the INQUIRY and the first (@marciot's) log does not? Can you attach the INQUIRY log for both commits, i.e. the commits with and without the bug? The first difference in these logs might reveal important information.

uweseimet, 2022-04-25:
@marciot What about setitng up a cross-compilation environment? That would even be faster than a Pi 4.

rdmark, 2022-04-25:
@uweseimet do you mean the section of the log that begins with ```[2022-04-25 09:12:43.449] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing Inquiry ($12)``` ?

uweseimet, 2022-04-25:
@rdmark Yes, that's whay I mean.

marciot, 2022-04-25:
> @marciot What about setitng up a cross-compilation environment? That would even be faster than a Pi 4.

@uweseimet: Is there documentation on how to do this? I tried compiling it under WSL Ubuntu and it failed spectacularly, so I gave up on trying that.

uweseimet, 2022-04-25:
@marciot There should be plenty of information in the internet, as cross-compilation with gcc is a standard approach for many environments. For the Pi I have never tried this myself, though.

marciot, 2022-04-25:
@uweseimet: What are you all using for development the ? I assume a Pi 4?

uweseimet, 2022-04-25:
@marciot Usually I use a regular Linux PC with the Eclipse IDE. I only use a Pi 4 when I need to test SCSI low level stuff. A lot of the relevant code does not deal with Pi-specific features, so that I can also test most changes on the PC.
For serious development, e.g. refactoring, one needs an IDE, and even the Pi 4 does not have enough resources (CPU and memory) for that.

uweseimet, 2022-04-25:
@rdmark @marciot After comparing the offending commit with the previous one I cannot find anything obviously wrong. Most changes do not affect the Daynaport emulation, and those that affect it essentially deal with INQUIRY and REQUEST SENSE, which is code also shared with other devices.
Can you please *both* confirm that the last working commit is da02eccb and the first non working commit is 1df7cdb1? I would like to ensure that you are both observing the same problem and that I am really comparing the right commits. Thanks a lot!

rdmark, 2022-04-25:
@uweseimet I tested again in my environment (Power Mac 8600 /  Mac OS 8.6) and can confirm that there's a breakage between those two commits.

Here's a log from a failure case, from when Mac OS has about halfway booted, until I open MacTCP and get an error that it cannot initialize the network interface.

```
[2022-04-25 13:38:06.334] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-25 13:38:06.335] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-25 13:38:06.335] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-25 13:38:06.335] [trace] virtual void SASIDEV::Command() CDB[0]=$12
[2022-04-25 13:38:06.335] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-25 13:38:06.335] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-25 13:38:06.335] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2022-04-25 13:38:06.335] [trace] virtual void SASIDEV::Command() CDB[4]=$01
[2022-04-25 13:38:06.335] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2022-04-25 13:38:06.335] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2022-04-25 13:38:06.335] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2022-04-25 13:38:06.336] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing Inquiry ($12)
[2022-04-25 13:38:06.336] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2022-04-25 13:38:06.336] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 13:38:06.336] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2022-04-25 13:38:06.336] [trace] void SASIDEV::Status() Status phase
[2022-04-25 13:38:06.336] [trace] void SASIDEV::Status() Status Phase $00
[2022-04-25 13:38:06.336] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 13:38:06.337] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2022-04-25 13:38:06.337] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2022-04-25 13:38:06.337] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 13:38:06.337] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 13:38:06.337] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 13:38:06.337] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2022-04-25 13:38:06.337] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2022-04-25 13:38:06.337] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-25 13:38:06.338] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-25 13:38:06.338] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-25 13:38:06.338] [trace] virtual void SASIDEV::Command() CDB[0]=$12
[2022-04-25 13:38:06.338] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-25 13:38:06.338] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-25 13:38:06.338] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2022-04-25 13:38:06.338] [trace] virtual void SASIDEV::Command() CDB[4]=$05
[2022-04-25 13:38:06.338] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2022-04-25 13:38:06.338] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2022-04-25 13:38:06.338] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2022-04-25 13:38:06.338] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing Inquiry ($12)
[2022-04-25 13:38:06.339] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2022-04-25 13:38:06.339] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 5
[2022-04-25 13:38:06.339] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2022-04-25 13:38:06.339] [trace] void SASIDEV::Status() Status phase
[2022-04-25 13:38:06.339] [trace] void SASIDEV::Status() Status Phase $00
[2022-04-25 13:38:06.339] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 13:38:06.339] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2022-04-25 13:38:06.339] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2022-04-25 13:38:06.340] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 13:38:06.340] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 13:38:06.340] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 13:38:06.340] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2022-04-25 13:38:06.340] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2022-04-25 13:38:06.340] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-25 13:38:06.340] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-25 13:38:06.340] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-25 13:38:06.341] [trace] virtual void SASIDEV::Command() CDB[0]=$12
[2022-04-25 13:38:06.341] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-25 13:38:06.341] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-25 13:38:06.341] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2022-04-25 13:38:06.341] [trace] virtual void SASIDEV::Command() CDB[4]=$24
[2022-04-25 13:38:06.341] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[... cut ...]
[2022-04-25 13:38:44.948] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2022-04-25 13:38:44.948] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 13:38:44.948] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 13:38:44.948] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 13:38:44.948] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2022-04-25 13:38:44.949] [trace] virtual void SCSIDEV::BusFree() Bus free phase
```

uweseimet, 2022-04-25:
@rdmark Can you please attach the corresponding log for the working commit? On first sight this part of the log

```
debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $08
[2022-04-25 13:38:23.640] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = SCSIDaynaPort; U = SASIDEV] Executing Read6 ($08)
[2022-04-25 13:38:23.640] [trace] virtual void SCSIDaynaPort::Read6(SASIDEV*) Control value 0, (0000), returning invalid CDB
[2022-04-25 13:38:23.640] [debug] Error status: Sense Key $05, ASC $24, ASCQ $00
[2022-04-25 13:38:23.640] [
```
looks suspicious to me.

uweseimet, 2022-04-25:
@rdmark No, this part of the log is probably fine. It means that a non-Daynaport driver (hard disk driver) tried to access the Daynaport, trying to read the root sector. This is most likely normal, it's just the hard disk driver's default behavior.
One can see that there are two drivers accessing the Daynaport, because one of them (the hard disk driver I guess) selects the LUN in a MESSAGE OUT phase (SCSI-2 compliant), the other does not request a MESSAGE OUT phase but uses the deprecated LUN bits in the CDB.

rdmark, 2022-04-25:
@uweseimet FYI this system is only using RaSCSI for the DaynaPort device in this configuration. The hard drive is a different SCSI device.

Here's a log from the good commit surrounding that Read6 message:

```
[2022-04-25 14:16:39.321] [trace] virtual void SCSIDEV::Execute() Execution phase command $08
[2022-04-25 14:16:39.322] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $08
[2022-04-25 14:16:39.322] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = SCSIDaynaPort; U = SASIDEV] Executing Read6 ($08)
[2022-04-25 14:16:39.322] [trace] virtual void SCSIDaynaPort::Read6(SASIDEV*) READ(6) command record=5 blocks=1
[2022-04-25 14:16:39.322] [trace] virtual int SCSIDaynaPort::Read(const DWORD*, BYTE*, uint64_t) Read maximum length 244, (00F4)
[2022-04-25 14:16:39.322] [trace] bool CTapDriver::PendingPackets() 0 revents
[2022-04-25 14:16:39.322] [trace] virtual int SCSIDaynaPort::Read(const DWORD*, BYTE*, uint64_t) No packet received
[2022-04-25 14:16:39.322] [trace] virtual void SCSIDaynaPort::Read6(SASIDEV*) ctrl.length is 6
[2022-04-25 14:16:39.322] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2022-04-25 14:16:39.322] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 6
[2022-04-25 14:16:39.323] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2022-04-25 14:16:39.323] [trace] void SASIDEV::Status() Status phase
[2022-04-25 14:16:39.323] [trace] void SASIDEV::Status() Status Phase $00
[2022-04-25 14:16:39.323] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 14:16:39.323] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2022-04-25 14:16:39.323] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2022-04-25 14:16:39.323] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 14:16:39.323] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
[2022-04-25 14:16:39.323] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2022-04-25 14:16:39.324] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2022-04-25 14:16:39.324] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2022-04-25 14:16:39.325] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-25 14:16:39.325] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-25 14:16:39.325] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-25 14:16:39.325] [trace] virtual void SASIDEV::Command() CDB[0]=$08
[2022-04-25 14:16:39.326] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-25 14:16:39.326] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-25 14:16:39.326] [trace] virtual void SASIDEV::Command() CDB[3]=$05
[2022-04-25 14:16:39.326] [trace] virtual void SASIDEV::Command() CDB[4]=$F4
[2022-04-25 14:16:39.326] [trace] virtual void SASIDEV::Command() CDB[5]=$C0
[2022-04-25 14:16:39.326] [trace] virtual void SCSIDEV::Execute() Execution phase command $08

```

uweseimet, 2022-04-25:
@rdmark The hard disk driver may initially access all devices, and this is what iti is actually doing. One can see that because he is using besides other commands READ CAPACITY, which ia disk-specific. All commands that use an IDENTIFY message are very likely those sent by the hard disk driver.
We definitely need the logfile from the working commit in order to compare the command sequence.

rdmark, 2022-04-25:
@uweseimet I see. Is there a good way to know when the hard drive activity has ended after a boot, by looking at the logs? I try to wait until the OS has seemingly finished starting up fully, but there may be residual hard disk activity that is not apparent.

uweseimet, 2022-04-25:
@rdmark Based on the usage of the IDENTIFY message it should be possible to identify (and discard) the hard disk driver commands, see my comment above.

rdmark, 2022-04-26:
@marciot Would you be able to take a look as well? The trick I'm using is to compile one rascsi binary before the commit, rename it to 'rascsi-good' or something, and then compile another binary that has the bug. That way you don't have to go back and forth compiling all the time.

I won't have much time tomorrow to spend on this due to a busy day at work...

marciot, 2022-04-26:
@rdmark: I started the first compilation when @uweseimet asked me to. It's finishing up the second compile now, about nine hours later. Commit da02eccb worked, commit 1df7cdb1 is not working.

uweseimet, 2022-04-26:
@rdmark @marciot I would like to ask again whether one of you can run a Mac tool that provides full information about the INQUIRY data returned by the Daynaport device for the two relevant commits.

marciot, 2022-04-26:
@uweseimet do you have any particular tool you recommend, or do I need to write my own?

uweseimet, 2022-04-26:
@marciot For the diffs you mean? In this case I was looking at the raw diffs created with 'git diff', but you can most likely use any IDE. I assume google will very likely find a bunch of other diff tools.

uweseimet, 2022-04-26:
Just for the record: As already mentioned I have issues with my setup which affects testing. Nevertheless I found a difference in the logs for the failure and success case.

Success:

```
[2022-04-26 10:30:08.658] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-26 10:30:08.658] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-26 10:30:08.658] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-26 10:30:08.658] [trace] virtual void SASIDEV::Command() CDB[0]=$12
[2022-04-26 10:30:08.658] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-26 10:30:08.659] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-26 10:30:08.659] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2022-04-26 10:30:08.659] [trace] virtual void SASIDEV::Command() CDB[4]=$25
[2022-04-26 10:30:08.659] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2022-04-26 10:30:08.659] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2022-04-26 10:30:08.659] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2022-04-26 10:30:08.659] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing Inquiry ($12)
[2022-04-26 10:30:08.659] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2022-04-26 10:30:08.659] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 37
[2022-04-26 10:30:08.659] [trace] virtual int GPIOBUS::SendHandShake(BYTE*, int, int) DELAYING for 100us after 6 bytes
```

Failure:
```
[2022-04-26 10:36:10.703] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2022-04-26 10:36:10.703] [trace] virtual void SCSIDEV::Selection() Initiator ID is 7
[2022-04-26 10:36:10.703] [trace] virtual void SASIDEV::Command() Command Phase
[2022-04-26 10:36:10.703] [trace] virtual void SASIDEV::Command() CDB[0]=$12
[2022-04-26 10:36:10.704] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2022-04-26 10:36:10.704] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2022-04-26 10:36:10.704] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2022-04-26 10:36:10.704] [trace] virtual void SASIDEV::Command() CDB[4]=$25
[2022-04-26 10:36:10.704] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2022-04-26 10:36:10.704] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2022-04-26 10:36:10.704] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2022-04-26 10:36:10.704] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing Inquiry ($12)
[2022-04-26 10:36:10.704] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2022-04-26 10:36:10.704] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 36
[2022-04-26 10:36:10.704] [trace] virtual int GPIOBUS::SendHandShake(BYTE*, int, int) DELAYING for 100us after 6 bytes
```

The Daynaport driver requests 37 INQUIRY bytes but after the problematic commit only 36 bytes are returned. It is legal to return less than the required amount of bytes, but either the Dayanport driver cannot deal with less than 37 bytes or there is an issue with the changes in the INQUIRY code that deals with the number of bytes to be returned.

uweseimet, 2022-04-26:
@rdmark @marciot Can you please change line 166 in devices/primary_device.cpp from

```
        vector<BYTE> buf = vector<BYTE>(0x1F + 5);
```

to
```
        vector<BYTE> buf = vector<BYTE>(0x1F + 6);
```

and see whether this resolves your issue? Even if this works it will not be the final resolution, because if the Daynaport requires 37 bytes this has to be dealt with in the Daynaport-specific code and not in the general INQUIRY code.

There are 36 standard INQUIRY bytes according to the SCSI specification (please correct me if I am wrong), but it looks as if the Daynaport driver expects one byte more, which can only be a vendor specific byte. In the working code this byte is 0.

uweseimet, 2022-04-26:
@rdmark @marciot In case the proposed fix above works, please also test the new fix_issue_750 branch, which is based on the develop branch.

rdmark, 2022-04-26:
@uweseimet The buffer size change made it work again on my Power Mac! I've tested both the quick and and the code infix_issue_750 branch. Great job figuring that out!

marciot, 2022-04-26:
@uweseimet: Compiling now. Will let you know if it resolves the issue. @rdmark, thank you for your efforts in troubleshooting this. I apologize that I was not able to help more. It's been a tough week for me.

marciot, 2022-04-26:
@uweseimet: It's all good! The network is properly initializing now!

closed 2022-04-26 by uweseimet