PiSCSI/piscsi #762: RASCSI Zip drive emulation issue: Roland SP808EX

lgagent29, 2022-05-06:
# Info

- Which version of Pi are you using: Pi2
- Which github revision of software: 22.2.1
- Which board version: 2.3
- Which computer is the RaSCSI connected to: Roland SP808EX

# Describe the issue
Hi, The only SCSI drives supported by the SP808EX are Zip drives. I have attepted operation with both the Zip 250 and 100mb images without succsess . The sp808ex detects the scsi port, however when attempting to format I get a "drive not ready" indicating the sp808ex belives the disk is not inserted. reseach has provided the sp808ex requires the ATAPI POI MODE 4 protocol for removable drives to read and write zip disks. Is this emulated in the current build of RASCSI? I have succsessfully tested RASCSI with the Roland DJ-70 MKII using the same zip images. the DJ-70 formatted both images correctly.

paulie-g, 2022-06-02:
Not connected with the development, but if you are correct that it requires ATAPI PIO Mode 4, it will not be in RASCSI. RASCSI emulates, unsurprisingly, SCSI. ATA (PATA aka IDE and SATA) is a different protocol over a different physical layer. Physical ATAPI Zip drives exist, they just have nothing to do, in interface terms, with RASCSI.

I'm not familiar with samplers, but a quick google indicates that the SP808 had a built-in Zip drive for 100MB zip disks, while the SP808EX had a built-in zip drive for 250MB zip disks. Presumably it is those, built-in drives, that are ATAPI. For the EX, you will want to use 250MB physical disks as 250MB zip drives write to 100MB zip disks very slowly. Those disks (and USB zip drives if you need to easily hook up to a modern computer) are very inexpensive on auction sites.

I did find a reference to a "Convert Disk" function on the SP808EX which would allow you to hook up RASCSI with a zip disk image on it and copy it over to the physical zip disk in the SP808EX. You will not be able to write to that RASCSI image though from what I can tell, and that's a SP808/EX limitation, not a RASCSI one. They were specifically designed so people wouldn't have to spend big $$$ on SCSI gear like they did on other samplers - zip disks were $20 MSRP at the time.

lgagent29, 2022-06-28:
Thanks for the info. I have two sp-808's, one EX and an original SP808 Ive upgraded with the EX software. Ive installed the MCD drive used in the blue AKAI's in the EX which works great. I was just hoping I could use the SCSI port on the OP1 board of the other SP808 for storage but alas not.

uweseimet, 2022-08-28:
@lgagent29 I'm afraid I am not sure whether this ticket is about SCSI or IDE/ATAPI drives. If it is not about SCSI this ticket should be closed because it does not affect RaSCSI. If it is about SCSI please provide a rasci trace log with the SCSI commands that are causing issues.

hsiboy, 2022-08-28:
I vote close. it was more of a general question.

uweseimet, 2022-08-28:
@hsiboy Thank you for your feedback. OK, I will close it.

closed 2022-08-28 by uweseimet