PiSCSI/piscsi #766: Various hard disk images are undetected and/or unbootable (depending on log level?)

unilock, 2022-06-17:
# Info

- Which version of Pi are you using: Raspberry Pi 4B, 4GB RAM, Raspberry OS Lite 64-bit (`Linux pi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`)
- Which github revision of software: **22.5.2 4cb2b64**
- Which board version: **2.4a**
- Which computer is the RaSCSI connected to: **Macintosh Plus, upgraded to 4MB RAM, I believe with the first revision (V2) of the hi and lo ROMs**

# Describe the issue

When attaching various hard disk images via RaSCSI with my Macintosh Plus, some won't boot (with various erroneous side-effects), and some aren't detected at all.
I've tested many images; I'll try to recall to the best of my ability which do and do not work.

.

"RaSCSI-Boot-6.0.8.hda" from [Macintosh Garden](https://macintoshgarden.org/apps/rascsi-68kmla-edition) (DL # 2) works when the log level is set to "trace". It gets stuck on the "Welcome to Macintosh" screen when set to any other log level.

When booting with log level "info", this is what appears in the log after getting stuck at the aforementioned screen:
```
Jun 17 08:26:41 pi RASCSI[506]: [2022-06-17 08:26:41.929] [info] Set log level to 'info'
Jun 17 08:26:41 pi RASCSI[506]: [2022-06-17 08:26:41.952] [error] Authentication failed
Jun 17 08:27:16 pi RASCSI[506]: [2022-06-17 08:27:16.299] [warning] Mode page data size exceeds reserved buffer size
```

On log level "debug":

```
Jun 17 08:29:30 pi RASCSI[506]: [2022-06-17 08:29:30.850] [info] Set log level to 'debug'
Jun 17 08:29:30 pi RASCSI[506]: [2022-06-17 08:29:30.867] [error] Authentication failed
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.493] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $08
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.493] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read6 ($08)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.530] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $08
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.530] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read6 ($08)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.596] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $08
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.596] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read6 ($08)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.642] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.642] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing Inquiry ($12)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.643] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.643] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing TestUnitReady ($00)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.645] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $25
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.645] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing ReadCapacity10 ($25)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.647] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.647] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.652] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.652] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.657] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.657] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.663] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.663] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.668] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.668] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.674] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.674] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.679] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:03 pi RASCSI[506]: [2022-06-17 08:30:03.679] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.132] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.132] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.663] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.663] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.669] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.669] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = ModePageDevice; U = SASIDEV] Executing ModeSense6 ($1A)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.669] [warning] Mode page data size exceeds reserved buffer size
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.676] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.676] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.686] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.686] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.694] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.694] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.701] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.701] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:04 pi RASCSI[506]: [2022-06-17 08:30:04.709] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
[... 89 lines cut ...]
Jun 17 08:30:06 pi RASCSI[506]: [2022-06-17 08:30:06.272] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:06 pi RASCSI[506]: [2022-06-17 08:30:06.272] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
Jun 17 08:30:06 pi RASCSI[506]: [2022-06-17 08:30:06.313] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Jun 17 08:30:06 pi RASCSI[506]: [2022-06-17 08:30:06.313] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = Disk; U = SASIDEV] Executing Read10 ($28)
```

This was the only image I had time to extensively test at the moment.

.

Some other images I've tried:

- Detected, bootable:
  - "Miniscribe M8425" created via web UI
[... cut ...]
  - "RaSCSI-BootstrapV3.hda" from [Macintosh Garden](https://macintoshgarden.org/apps/rascsi-68kmla-edition)
- Undetected:
  - Any other image(?) created via RaSCSI's web UI
  - Any image created by Basilisk II
- When powering on the Mac with any large(?) image attached, it shows a black screen, produces static noise, then shows static on its screen. One such image would be the Apple Legacy Recovery CD (as a disk image, [found here](https://mega.nz/folder/8hA3AQCJ#pWUq92L70yDXlogy9lk5Dg/file/gsRCxDxC) - again, unknown source).

.

Let me know what more information I can provide.

rdmark, 2022-07-02:
@unilock The Mac Plus is known to have issues booting from a SCSI drive with particular hard disk drivers. See https://github.com/akuker/RASCSI/issues/40. Try initializing the drive image with various tools/drivers to figure out what works in your setup.

joeventura1, 2022-07-11:
> @joeventura1 See wiki: https://github.com/akuker/RASCSI/wiki/Drive-Setup#initializing-a-new-disk-on-classic-mac-os

Thanks Silverlining seems to have done the trick

rdmark, 2022-07-29:
Closing, as this is a known limitation of the Plus, with known workarounds.

closed 2022-07-29 by rdmark