PiSCSI/piscsi #769: VAXstation 3100 no longer boots on latest release

Pacjunk, 2022-07-05:
# Info

- Which version of Pi are you using: Zero W
- Which github revision of software: Release v22.05.02
- Which board version: FullSpec
- Which computer is the RaSCSI connected to: VAXstation 3100.

# Describe the issue

Changes to the modesense(6) code (#699) is now preventing the VAXstation from booting. _Previous versions worked fine_. The latest version does not allow page 0, and this is required by the boot ROM. The latest version works fine using the **operating system** driver, but not the **boot driver** in ROM. Attached is relevant part of trace:
```
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.232] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=2 (with device)
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.232] [trace] virtual void SCSIDEV::Selection() Initiator ID is 6
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.232] [trace] virtual void SASIDEV::Command() Command Phase
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.232] [trace] virtual void SASIDEV::Command() CDB[0]=$1A
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.232] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.232] [trace] virtual void SASIDEV::Command() CDB[2]=$00
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.232] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] virtual void SASIDEV::Command() CDB[4]=$0C
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] virtual void SCSIDEV::Execute() Execution phase command $1A
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = ModePageDevice; U = SASIDEV] Executing ModeSense6 ($1A)
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] int ModePageDevice::AddModePages(const DWORD*, BYTE*, int) Requesting mode page $00
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] int ModePageDevice::AddModePages(const DWORD*, BYTE*, int) Unsupported mode page $00
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [debug] Error status: Sense Key $05, ASC $24, ASCQ $00
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] virtual void SCSIDEV::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] void SASIDEV::Status() Status phase
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] void SASIDEV::Status() Status Phase $02
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.233] [trace] void SASIDEV::MsgIn() Starting Message in phase
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.234] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.234] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.234] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Jul 04 16:30:59 raspberrypi RASCSI[432]: [2022-07-04 16:30:59.234] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)

```
The boot ROM tries this 3 times, then gives up. Since RASCSI supports retro systems, shouldn't it allow things like this (even though it may not be part of the latest spec)?

Thanks,

Pacjunk, 2022-07-05:
Very strange - When I boot bluescsi, the system requests page $3F (all pages). Don't know why it requests $00 for RASCSI. Maybe it is getting an unexpected answer from the preceding Inquiry and/or TestUnitReady commands?

I think the change to disallow page 0, is causing the boot to fail but Bluescsi also disallows it and works fine because if gets a 3F. rather than 00. More investigation required...

Pacjunk, 2022-07-05:
The only thing I've found so far is that RASCSI returns SCSI-2 for the Inquiry, whereas Bluescsi returns SCSI-1.

rdmark, 2022-07-08:
@Pacjunk So just to be clear: Also the previous stable versions of the RaSCSI software responded "SCSI-2" to the same inquiry from the VAXstation? And that the only difference is how RaSCSI reacted to page code $0, from what you can see?

rdmark, 2022-07-08:
Looking at the code, there is actually logic for responding SCSI_1_CCS to an INQUIRY from a non-SCSI-2 device:

https://github.com/akuker/RASCSI/blob/348b5dd026804f4abab99d44808a94e79e82838b/src/raspberrypi/devices/primary_device.cpp#L178

Next, to figure out how the scsi_level is detected, and why this controller is misidentified.

Pacjunk, 2022-07-08:
> @Pacjunk So just to be clear: Also the previous stable versions of the RaSCSI software responded "SCSI-2" to the same inquiry from the VAXstation? And that the only difference is how RaSCSI reacted to page code $0, from what you can see?

The VAXstation ROM code sends an inquiry to the device. If it returns SCSI-1, then it sends a modesense with page code $3F. If successful, the boot continues. If the device returns SCSI-2, then for some reason the VAXstation then requests a modesense with page code $00. In the past RASCSI ignored this and returned a success status (0), so the VAXstation has happy. After the latest changes page code $00 returns an error (2) and the VAXstation will not boot due to the error. It retries a few times then stops.

This only occurs with the boot ROM (and I assume because it is dated prior to the SCSI-2 standard). Once it boots, the OS driver takes over and this driver does things correctly (It is dated 7 or 8 years later). If I boot off a real HDD, I can then mount and use RASCSI volumes without issue.

Now the changes to RASCSI are appropriate and page 0 **should** be rejected. Just have to sort out how the SCSI version is determined, and if it is fixed (It is from what I could see), then how do we easily provide a method to set it?

A bunch of the predefined drive types that RASCSI provides are SCSI-1 only, so that is why I was suggesting adding something to the properties files, then RASCSI could reply with the appropriate version for the device that it is emulating.

Thanks,

Pacjunk, 2022-07-08:
OK, done some more reading. There are two fields:

"ANSI-approved version"  : 0 for no standard, 1 for SCSI-1, 2 for SCSI-2
"Response data format"    : 0 for strictly SCSI-1, 1 for current with backward compatibility, 2 for strictly SCSI-2

> A response data format value of zero indicates the INQUIRY data format is as specified in SCSI-1. A response data format value of one indicates compatibility with some products that were designed prior to the development of this standard (i.e. CCS). A response data format value of two indicates that the data shall be in the format specified in this International Standard. Response data format values greater than two are reserved.

The code in RASCSI scsihd.cpp that I mention above, sets the ansi version to 2. The code snippit you give appears to set the response format to 2 as well. From my understanding this means the emulated device is SCSI-2 and is **not** backward compatible. Going back to bluescsi (because it is quick and easy for me to test!) I set the ansi version to 2, and response to 1 - and it worked!  Setting to 2 and 2 (same as RASCSI) failed. Default for bluescsi is 1 and 1. I note that the default for scsi2sd is 2 and 1.

Need some SCSI expert to check this, but there seem to be two options:

A: Set the response data format value to 1 across the board
B: Allow the SCSI version to be changed somehow, preferably per disk.

I need to check a physical known compatible (Digital branded) SCSI-2 drive tomorrow to see what values it has set.

uweseimet, 2022-07-09:
@rdmark It was just a matter of time until it turned out that the INQUIRY SCSI level should be configurable. One new parameter for setting the SCSI/SPC level (valid values 0-8 I think in order to cover all existing specifications) should be sufficient.The INQUIRY response code can be derived from the level because it is not really independent from it. I think the current code already derives the response code from the level, so just making the SCSI/SPC level configurable should be all that´s needed.
I won´t have time to address this soon, but somebody else can do this, of course. It´s no rocket science. The current code for setting the INQUIRY device name is a good guideline. Besides adding the new setting to rascsi and rasctl the remote interface has to be extended (in a backwards compatible fashion).

Pacjunk, 2022-07-09:
Would it really do any harm to just set the response format to 1?  Given that RASCSI hard codes the level to SCSI-2, and the majority of the users are probably using SCSI-1 machines, doesn't it make sense to enable backward compatability?

uweseimet, 2022-07-09:
@Pacjunk Since the most elaborate and common SCSI level is SCSI-2, this should be the default. Otherwise your quite exotic platform would work  and less exotic ones might fail.
Until somebody makes the level configurable you can just change it in your local code.

Pacjunk, 2022-07-09:
I'm not saying that we should change the default scsi level. 2 is appropriate. I'm asking if we should change the response format field to 1 for compatibility?  SCSI2SD seems to set these 2&1 values.  Unfortunately I don't have other systems to test on. I agree that my machine has some SCSI anomalies that later machines did not have due to its age. Sure I can change my version of the code to work, I just wanted to raise the issue.  Thanks,

Pacjunk, 2022-07-10:
OK, now for the next problem!!  I changed my local copy to set the response format field to 1, and the system now generates page code $3F instead of $00.

```
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.642] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=2 (with device)
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.642] [trace] virtual void SCSIDEV::Selection() Initiator ID is 6
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.642] [trace] virtual void SASIDEV::Command() Command Phase
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.642] [trace] virtual void SASIDEV::Command() CDB[0]=$1A
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.642] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.642] [trace] virtual void SASIDEV::Command() CDB[2]=$3F
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] virtual void SASIDEV::Command() CDB[4]=$0C
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] virtual void SCSIDEV::Execute() Execution phase command $1A
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = ModePageDevice; U = SASIDEV] Executing ModeSense6 ($1A)
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] int ModePageDevice::AddModePages(const DWORD*, BYTE*, int) Requesting mode page $3F
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [warning] Mode page data size exceeds reserved buffer size
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] virtual void SCSIDEV::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] void SASIDEV::Status() Status phase
Jul 10 14:50:18 raspberrypi RASCSI[432]: [2022-07-10 14:50:18.643] [trace] void SASIDEV::Status() Status Phase $02
```

What happens now is that in the request it specifies a length of $0C, which is only enough to hold the mode parameter block, not all the pages, and it looks like the #699 changes trap this mismatched length, generates a warning message "*Mode page data size exceeds reserved buffer size*", and then generates an error 2. So the effect is the same - the system errors out and refuses to boot. Originally it failed because page code 0 was rejected, now it fails because the requested length is incorrect.

I checked the bluescsi code once again, and it just truncates the page code data to the requested size (which is enough for the mode parameter block) and everything is happy.

Is it intended that this "warning" generates an error status? (in which case it is an error, rather than a warning). Should it be handled in a different way?

Thanks,

rdmark, 2022-07-29:
@uweseimet Do you have an idea why the warning leads to an error state, as per Pacjunk's observation above?

uweseimet, 2022-07-29:
@rdmark Yes, it's essentially a code optimization.
All in all in order to address this issue an additional temporary vector (i.e. a buffer with dynamic size) to hold all result data would have to be created, and at the very end just the requested number of bytes would have to be returned.

uweseimet, 2022-08-18:
@Pacjunk There is a fix_partial_mode_pages branch now which should fix the issue with incomplete mode pages reported by you on July 10th. Can you please test this change in your environment?

uweseimet, 2022-08-18:
@Pacjunk The [feature_google_test branch contains the change from the fix_partial_mode_pages branch and some unit tests (based on GoogleTest) for this change. I don't know yet whether there is a GoogleTest package for the Pi, but this probably does not matter because in case I continue with adding unit tests (so far there haven't been any) I will most likely only add platform-independent tests that can be executed on a rregular PC anyway.
As expected, one of the unit tests fails with the develop branch, but succeeds with the feature branch ;-).

Pacjunk, 2022-08-19:
@uweseimet The feature_google_test branch just gives me the following error during build...
```
./test/rascsi_test.cpp:3:10: fatal error: gtest/gtest.h: No such file or directory
 #include <gtest/gtest.h>
          ^~~~~~~~~~~~~~~
```
so I tried the fix_partial_mode_pages branch, but I have to go and hack the code to get around the original problem (system requesting page code 0, due to getting scsi-2 response, and page code 0 no longer being supported). I will probably have an answer for you by tonight (my time). Thanks for working on this.

Pacjunk, 2022-08-19:
Nope, didn't work!  After I hacked the code to get around the page code 0 issue, the host requested page $3F and length $0C (as expected), but this still generated an error message. Log is below:

```
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=2 (with device)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SCSIDEV::Selection() Initiator ID is 6
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SASIDEV::Command() Command Phase
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SASIDEV::Command() CDB[0]=$1A
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SASIDEV::Command() CDB[2]=$3F
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SASIDEV::Command() CDB[4]=$0C
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.105] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] virtual void SCSIDEV::Execute() Execution phase command $1A
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = ModePageDevice; U = SASIDEV] Executing ModeSense6 ($1A)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] int ModePageDevice::AddModePages(const DWORD*, BYTE*, int) Requesting mode page $3F
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] virtual void SCSIDEV::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] void SASIDEV::Status() Status phase
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] void SASIDEV::Status() Status Phase $02
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] void SASIDEV::MsgIn() Starting Message in phase
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.106] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=2 (with device)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SCSIDEV::Selection() Initiator ID is 6
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SASIDEV::Command() Command Phase
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SASIDEV::Command() CDB[0]=$03
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.107] [trace] virtual void SASIDEV::Command() CDB[2]=$00
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [trace] virtual void SASIDEV::Command() CDB[4]=$08
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [trace] virtual void SCSIDEV::Execute() Execution phase command $03
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [debug] bool Dispatcher<T, U>::Dispatch(T*, U*) [with T = PrimaryDevice; U = SASIDEV] Executing RequestSense ($03)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [trace] virtual std::vector<unsigned char> PrimaryDevice::RequestSense(int) Status $00, Sense Key $00, ASC $01
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [trace] void SASIDEV::DataIn() Going into Data-in Phase
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.108] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 8
[... cut ...]
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.109] [trace] virtual void SCSIDEV::Send() Sending handhake with offset 0, length 1
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.109] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.109] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Aug 19 17:02:36 raspberrypi RASCSI[436]: [2022-08-19 17:02:36.109] [trace] virtual void SCSIDEV::BusFree() Bus free phase
```

uweseimet, 2022-08-19:
@Pacjunk I may not have been precise enough on the nature of my change: It was exclusively about the "Mode page data size exceeds reserved buffer size" warning and about rascsi not returning any mode page data when this warning is displayed. The warning is gone and as many mode page data as requested by the allocation length are now returned. This is also confirmed by the unit tests.

Regarding mode page 0 there was no change. Mode page 0 is vendor-specific, so which mode page data should rascsi returnn? Each device can have its own set of data for this mode page, but rascsi cannot dynamically decide which set of data to return for a particular device in order to satisfy a particular computer.

uweseimet, 2022-08-19:
@Pacjunk Regarding GoogleTest, it appears not to be available for Raspberry Pi OS, but maybe it is for Pis with Ubuntu. But as already mentioned, running unit tests on a regular PC is probably the better choice, because the Pi is not such a good development environment, and compiling tests in addition to the regular sources makes everything even slower.

Pacjunk, 2022-08-19:
> @Pacjunk I may not have been precise enough on the nature of my change: It was exclusively about the "Mode page data size exceeds reserved buffer size" warning and about rascsi not returning any mode page data when this warning is displayed. The warning is gone and as many mode page data as requested by the allocation length are now returned. This is also confirmed by the unit tests. PR for this change is #793.
> 
> Regarding mode page 0 there was no change. Mode page 0 is vendor-specific, so which mode page data should rascsi returnn? Each device can have its own set of data for this mode page, but rascsi cannot dynamically decide which set of data to return for a particular device in order to satisfy a particular computer.

Yes, but as I say, I hacked the code so that my system doesn't request page 0 (this is caused by the response code being set to 2, instead of 1). If you look at the trace log you can see the system is requesting page code $3F (which is valid) with the limited length that your fix is supposed to address, but an error is still generated somewhere. Unfortunately the trace log doesn't give any more detail.

uweseimet, 2022-08-19:
@Pacjunk It's suspicious that the error codes are 0 except for the Additional Sense Key, which is $01. I don't think this makes any sense. Have you already checked the rascsi code? This might be caused by a typo somewhere.

uweseimet, 2022-08-19:
@Pacjunk Regarding the SCSI level/response code format I have a question: Which extension does your image file have? .hds I assume, i.e. a generic SCSI drive image?
I am asking this because instead of making the SCSI level and response code format configurable it may be better to just introduce a new .hd1 image file type for generic SCSI-1 drives. .hd1 would set SCSI level and response code format to 1, which would be fine for platforms with only SCSI-1 but no SCSI-2 support. This implicit configuration, based on the extension, avoids to introduce two new configuration settings for rather exotic platforms that cannot deal with SCSI-2. Otherwise, with additional explicit settings, much more code would have to be changed. Not just in rascsi itself but also in all clients (rasctl, web interface, RaSCSI control app). The clients would have to offer a means to configure these values, but for most users this would be confusing, because the settings are usually not needed and require expert knowledge in order to use them correctly.
From your perspective, would a new .hd1 extension for generic SCSI-1 drives be a working solution?

Pacjunk, 2022-08-19:
> @Pacjunk It's suspicious that the error codes are 0 except for the Additional Sense Key, which is $01. I don't think this makes any sense. Have you already checked the rascsi code? This might be caused by a typo somewhere.

Well, it's not my code, so it would take me a while to try and trace it out. I still have trouble getting my head around the way all the modules for RASCSI hang together.

Pacjunk, 2022-08-19:
> @Pacjunk Regarding the SCSI level/response code format I have a question...

From my reading of the spec, the response code doesn't indicate the SCSI version. From memory it says that '2' means that the device adheres to the selected standard and nothing else (ie SCSI-2 in this case), and '1' means that it adheres to the selected standard **but** allows backward compatibility. Can you have a look at the spec to verify?

Now the issue with my system, does not occur because of the SCSI level. It can be set to 1 or 2. What changes the result is the setting of the response code. If set to 1 (backwards compatibility) everything works fine. If it is set to 2, then the system erroneously generates the request for page code 0 (which it normally wouldn't do).

Would it be such a problem to set the response code to 1 to allow for compatibility? The SCSI level does **not** need to be changed. I realise that testing would be required on other platforms, which I unfortunately do not have.

Pacjunk, 2022-08-19:
> @Pacjunk Well, the more exotic your hardware, the more you may have to get to know the code if you want to your hardware to be supported ;-)

I don't think my hardware is doing anything exotic. Its just a modesense command, page $3F, length $0C. Its up to RASCSI what it does with that. I will try to have a look at the code. I don't use an IDE, I'm not an RASCSI developer :-)

uweseimet, 2022-08-19:
@Pacjunk The natural response format of a SCSI-2 drive is the SCSI-2 response format. Using SCSI-1 responses for SCSI-2 drives may cause issues with other platforms, which is a risk that cannot be taken. You have a quite exotic platform, which cannot be used a reference platform for the implementation.
Currently (you might want to verify from the code) the response format is derived from the SCSI level. If the SCSI level is SCSI-2 (or a newer SCSI level) device, the response format is SCSI-2. If the SCSI level is SCSI-1, the response format is SCSI-1. This means that if want the response format to be SCSI-1, you automatically get it if the SCSI level is also SCSI-1.

uweseimet, 2022-08-19:
> I don't think my hardware is doing anything exotic. Its just a modesense command, page $3F, length $0C. Its up to RASCSI what it does with that. I will try to have a look at the code. I don't use an [...]

You may have to become one. I also did not implement the original rascsi code, and had to spend time to get familiar with it.
When I say "exotic" I mean your hardware. It is definitely more exotic than a Mac, Amiga or Atari ;-). And expecting SCSI-1 responses from SCSI-2 drives is IMHO also exotic. My guess is that your hardware simply expects a SCSI-1 drive, where the natural response format is SCSI-1.

Pacjunk, 2022-08-19:
Yes, I've seen the code as this is what I hack to get to the next stage of testing the modesense length. The issue only seems to be with the ROM code on that machine. The OS driver seems to work fine, so I suppose it is an edge case.

I have no issue with changing the image extensions if that makes it easy but it would be nice if the GUI had some way of setting this when creating the drive (I realise this is not your area). The web interface code would probably need checking over anyway if this was to be done.

uweseimet, 2022-08-19:
@Pacjunk As already mentioned, forcing even more configuration options on the users is IMO not such a good idea. Most users definitely do not know what SCSI levels, SCSI response format versions and their relationship is. The clients (it's not just the web UI, as already pointed out, but also others) should not become even more complicated.

Pacjunk, 2022-08-19:
> And expecting SCSI-1 responses from SCSI-2 drives is IMHO also exotic. My guess is that your hardware simply expects a SCSI-1 drive, where the natural response format is SCSI-1.

I just experimented until I found a combination that worked, and that was that the SCSI level didn't matter, but the response level did. What the ROM code is actually doing is beyond my intelligence. Setting both to 1 works. I was just looking for a minimal change, but if you think changing the response code is too much of a risk, then we need to look at alternatives as you have suggested.

uweseimet, 2022-08-19:
@rdmark I may repeat myself, but anyway: If just changing the SCSI level to 1 resolves your issue (because it implicitly also changes the response format code to 1), this would be best. Then we may have a simple soluation (.hd1 image file). No other options and no client changes (UI etc.) would be required, and the amount of code changes would be minimal. This is why I favor this approach.

Pacjunk, 2022-08-19:
> @Pacjunk As already mentioned, forcing even more configuration options on the users is IMO not such a good idea. Most users definitely do not know what SCSI levels, SCSI response format versions and [...]

That's OK, I'm happy to manually rename my images. Any GUI option wouldn't refer to the response formats, just set '1' or '2' for SCSI in general.

uweseimet, 2022-08-19:
@Pacjunk Well, it is sometimes a tradeoff whether to introduce an additional extension or an additional option in the clients. Regarding the latter it is important to remember that the server/client interface is an open interface, and changing such an interface does not just mean changes on the server side, but may also mean changes on the client side. There are currently at least 3 clients. I say "at least" because you never know wether there are more. Adding functionality can usually be done in a backwards compatible fashion. But if there is no pressing need for such a change (e.g. a bugfix or a relevant new feature), or a better alternative, one should try to do without it.

Pacjunk, 2022-08-19:
My only concern is whether things like the linked properties files will still work if the image extensions are changed. I suppose there is one way to find out!

One interesting thing I have seen regarding this is that if I use a real SCSI-2 hard drive (ie both level and response are 2), then my system works fine! The only thing I can think of is that the hard drive in question is **not** ignoring page code 0. The host only asks for the 12 bytes of the mode parameter block anyway, which is provided with every modesense request regardless of the page. Setting the SCSI level was only my next option after the page code 0 option was blocked. (ie it was not a problem before page 0 was disallowed)

Pacjunk, 2022-08-19:
That I suppose is another option to consider. If page code 0 is requested, then just return the mode parameter block. After that my system doesn't really care about the scsi level, so no changes to image extensions required.

uweseimet, 2022-08-19:
@Pacjunk Whatever is returned as page code 0 must be valid based on the SCSI specification. Otherwise platforms that retrieve page 0 but have a fallback if page 0 is not avalable would break. From other tickets we can see that there actually are platforms that react dynamically on the outcome of SCSI commands. Often this is related to optimizations. Typical (existing) example: A driver uses REPORT LUNS in order to find which LUNs are available without checking for each LUN explicitly. If REPORT LUNS is not available the driver will check for each LUN separately as a fallback strategy.

Pacjunk, 2022-08-19:
Yes, understand that. The physical HDD that I have must be returning something for page code 0, otherwise my machine would not work. Who knows what it is :-)  As you say, its vendor specific (seagate, quantum etc).

Looking at the code now, trying to work out what it all means!  Did you test the case of length 12?  In this case addmodepages is called with a maximum length of 0. Haven't determined if this causes a problem though.

Pacjunk, 2022-08-19:
In Disk::ModeSense6, there is code:
```

	int pages_size = super::AddModePages(cdb, &buf[size], length - size);
	if (!pages_size) {
		return 0;
	}
	size += pages_size;

	if (size > 255) {
		SetStatusCode(STATUS_INVALIDPRM);
		return 0;
	}
```
length - size will be 0 in this case.  ModePageDevice::AddModePages will return 0 as well. So I assume the "if (!pages_size) return 0;" bit will cause an error status, but it does not set a status code like the next test does. The modesense command seems to return a non-zero value for success.

I think this is the problem area, but how to fix it I wouldnt know!

Pacjunk, 2022-08-19:
Same code in modesense10 section as well. I can't see that it will do any harm to remove:

```
if (!pages_size) {
		return 0;
	}
```

uweseimet, 2022-08-19:
> Looking at the code now, trying to work out what it all means! Did you test the case of length 12? In this case addmodepages is called with a maximum length of 0. Haven't determined if this causes a problem though.

@Pacjunk Whether the unit test uses length 12 or 1 does not matter. In order to test partial pages any value smaller than the actual number of available bytes is fine.
```
EXPECT_EQ(hd_device.AddModePages(cdb, buf, 12), 12);
```
passes just like
```
EXPECT_EQ(hd_device.AddModePages(cdb, buf, 1), 1);
```

> The modesense command seems to return a non-zero value for success.

You mean it reports an error condition for success? I don't think so, because in this case MODE SENSE would not work at all.

I wonder where there seems to be an error condition for page code $3F in your log. I don't think the fix_partial_mode_pages can ever return 0 when you request $0C bytes. I suggest that you add debug code in order to analyze what's actually going wrong.

> Same code in modesense10 section as well. I can't see that it will do any harm to remove:

I definitely don't want the rest of the code to be executed. If the rest of the code is extended later, not returning early might result in issues. No need to take risks here.

rdmark, 2022-08-19:
> @akuker @rdmark Let me point out here that #729 not being addressed makes everything more complicated, because currently there is a mixture of SASI and SCSI error handling. As far as I can see it's obvious that there is no interest in SASI support.

@uweseimet Are you interested in addressing #729? We've informed the community about our intention to deprecate SASI in the future, so we're ready to execute on that plan whenever someone takes the initiative to do the coding and cleanup work.

uweseimet, 2022-08-19:
@rdmark Yes, I would most likely start working on this ticket. This will most likely trigger a lot of code changes, and may required several steps with separate PRs.
What about starting with this ticket after the next release?

rdmark, 2022-08-19:
@uweseimet This sounds like a reasonable roadmap for these changes.

Pacjunk, 2022-08-19:
@uweseimet I've shown you where it goes wrong. The code builds the mode parameter block which is 12 bytes long, it then adds all the mode pages ($3F), then trims them all off again as the host only requested 12 bytes. pages_size is equal to 0, so the IF statement exits the procedure without returning the mode parameter block part. Why exit here when you have valid data to return?  All the "work" has been done at this stage, so I can't see that removing that test will cause any problems. All it will do is to allow the code to return the valid mode parameter block.

Pacjunk, 2022-08-20:
> Same code in modesense10 section as well. I can't see that it will do any harm to remove:
> 
> ```
> if (!pages_size) {
> 		return 0;
> 	}
> ```

@uweseimet I've tested my fix and it works fine. System boots and operates correctly.

Up to you what you want to do about it!

uweseimet, 2022-08-20:
@Pacjunk Now I think I see your point, please confirm: You say that in this call:

```
int pages_size = super::AddModePages(cdb, &buf[size], length - size);
```

"length - size" evaluates to 0 when the allocation length is $0C, that's why AddModePages returns 0, and you do not want a return value of 0 to raise an error.
Correct me if I am wrong, but if you omit the test no error would be raised anymore when you request a non-existing page. Only the header would be returned and success would be reported. Is this compliant with the SCSI specification?

Pacjunk, 2022-08-20:
> Correct me if I am wrong, but if you omit the test no error would be raised anymore when you request a non-existing page. Only the header would be returned and success would be reported. Is this compliant with the SCSI specification?

@uweseimet No that case is trapped by code in mode_page_device.cpp

```
	// If no mode data were added at all something must be wrong
	if (pages.empty()) {
		LOGTRACE("%s Unsupported mode page $%02X", __PRETTY_FUNCTION__, page);
		SetStatusCode(STATUS_INVALIDCDB);
		return 0;
	}
```

But - This returns 0 to flag the error, so that is what the other code is trapping for I suppose. But this can also occur in my case where pages **are** valid, but length is 0.  Maybe the above error condition needs to pick another value for flagging the error.
e.g. the above code can return -1, then the test I removed could check for -1 ??

uweseimet, 2022-08-20:
@Pacjunk We are almost at a point where exceptions might be better to propagate errors instead of some (magic) integer values.
But your suggestion appears fine to me. For now (in order not yet to create a new branch) I added this change to the feature_google_test branch. I have not yet tested it, but I suggest that you have a look at my change and run a test with your setup.

Pacjunk, 2022-08-20:
Well I see the problem now is that the function can return 0 for two reasons. Either an error, or a request for 12 bytes. So if the request for 12 bytes is to be allowed, then the flagging of the error must be different. I just suggested -1 as a machine should never request less than 12 bytes unless it is really broken. Maybe it should be -13 or -maxint (if that is a thing). Anyway...

I think your code has got it the wrong way around. It should be:

```
	// If no mode data were added at all something must be wrong
	if (pages.empty()) {
		LOGTRACE("%s Unsupported mode page $%02X", __PRETTY_FUNCTION__, page);
		SetStatusCode(STATUS_INVALIDCDB);
		return -1;
	}
```
and then:

```
	if (pages_size==-1) {
		return 0;
	}

```

maybe there needs to be a symbol defined (e.g. errorflag) and just test for that. Might make it more readable.

uweseimet, 2022-08-20:
@Pacjunk Indeed, this simple change would not work, I just reverted it in the branch. But we definitely must not starting using magic values like -13.
IMO there is a general issue with how the error handling is organized between controllers and devices, and as we see from this case particularly propagating errors is an issue. I created and resolve some tickets in the past in order to fix some other issues related to the responsibilities of controllers and devices, and I think we need a more general change for a reliable solution.

The setting of the SCSI error codes is currently done in various places in the Disk class and derived classes. Error codes are set on various call hierarchy levels, either directly or indirectly like in our case, triggered by some function return values. If a function returns an error code one of the callers in the hierarchy is responsible for taking the right action, i.e. setting the SCSI error code and/or returning a flag indicating that there was an error, or at least ensuring that no further code is executed.
By introducing an exception that signals SCSI errors and their Sense Key and ASC we could simplify the error handling in the Disk and drived classes. In case of an error this exception can be caught in the PrimaryDevice::Dispatch() method, and in this method controller->Error() is called with the values propagated by the exception.

This would also simplify code constructs like this, which you find quite a lot:

```
void PrimaryDevice::TestUnitReady(SASIDEV *controller)
{
	if (!CheckReady()) {
		controller->Error();
		return;
	}

	controller->Status();
}
```

would essentially become

```
void PrimaryDevice::TestUnitReady(SASIDEV *controller)
{
	CheckReady();

	controller->Status();
}
```
because CheckReady() throws an exception in case of an error.

What's your opinion this? It's a major change, of course, but there are a lot of code quality issues like that which required major changes in order to get the code into a better shape.

Pacjunk, 2022-08-20:
As you say, this stuff is riddled everywhere!  In the above example:
```
	if (pages_size==-1) {
		return 0;
	}
```
it is checking for one flag, and then setting another flag for the next level up!  How far do we go with this?

As to your proposal, it is above my level of knowledge to comment!  (I'm not great at C programming)

How do the exceptions know where to return control? e.g. how do you ensure flow moves to the next scsi phase.

As this change will likely take months to implement, I think I will just hack my local copy with these changes, just to get things going for me, and leave the heavy lifting to you  :-)

uweseimet, 2022-08-20:
@Pacjunk This change will not take months, but maybe weeks ;-). So till then I also think you should use a local copy.
Regarding bus phases nothing will change, because just like now after Error() was called while dispatching a command the next phase will automatically be the status phase. This is not affected by introducing an exception, but there will be much less (maybe one ony) code locations where Error() is called, namely the location where the exception is caught. All other locations will raise this exception.

uweseimet, 2022-08-20:
I created https://github.com/akuker/RASCSI/issues/795 as a separate ticket for exception-based error handling.

@rdmark Based on the discussion above it turned out that major error handling changes are required in order to get closer to a solution for this ticket. Removing SASI support will reduce the effort for these changes.
In order not to lose too much momentum I suggest not to wait too long for the next release. What about having a 22.08 release at the end of this month. It will not contain major changes but some user-visible improvements have already been committed as far as I can tell.
I would also like https://github.com/akuker/RASCSI/issues/794 to be part of this release, in order to at least have a starting point for further unit tests.

Pacjunk, 2022-08-20:
I have created a fork with my workarounds and will build off that for the time being.

uweseimet, 2022-08-20:
@Pacjunk Sounds good. It may make sense to also create a new ticket as a replacement for this one. This new ticket could describe the (remaining) issues from nowadays perspective. The current ticket makes it hard to find out what the current/remaining problems are because one would have to read and understand the numerous comments.

rdmark, 2022-08-21:
@uweseimet Agreed, I think we have a good changeset lined up to warrant a new release. I spoke to @akuker and we decided to aim to tag the release next weekend.

Pacjunk, 2022-08-21:
This thread has become far too long to easily understand, so I have created a new issue #799 with a summary of the issues and possible solutions.

Pacjunk, 2022-08-21:
Continued in #799

closed 2022-08-21 by Pacjunk