PiSCSI/piscsi #773: Move .rascsi_secret to config directory

nucleogenic, 2022-07-29:
A config directory was introduced in PR https://github.com/akuker/RASCSI/pull/299 at ` $HOME/.config/rascsi`.

This issue proposes `.rascsi_secret` is relocated to the config directory for consistency and improved portability.

If the file is moved from $HOME, is there a tangible benefit to it remaining a dotfile?

rdmark, 2022-07-29:
Thanks for reporting! I had a look at the code, and contrary to my assessment in the Discord thread, it turned out this is a simple fix in easyinstall.sh. I have a fix lined up that I'm testing now.

The one argument I can think about for keeping it as a dotfile is to distinguish it from all the json config files and properties files that might litter .config/rascsi

rdmark, 2022-07-29:
https://github.com/akuker/RASCSI/pull/774

rdmark, 2022-07-29:
Merged

closed 2022-07-29 by rdmark