PiSCSI/piscsi #775: Slow upgrades via easyinstall.sh

nucleogenic, 2022-07-30:
# Info

- Which version of Pi are you using: Raspberry Pi Zero 2 Rev 1.0
- Which github revision of software: [348b5dd](https://github.com/akuker/RASCSI/commit/348b5dd026804f4abab99d44808a94e79e82838b)
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: N/A

# Describe the issue

Upgrading RaSCSI via easyinstall.sh takes a long time (>30 mins on RPi Zero 2) largely due to the compilation step.

This could be considered a barrier to users upgrading their RaSCSI installation.

One approach to this issue could be the distribution of binaries, which been mentioned as part of https://github.com/akuker/RASCSI/issues/740.

uweseimet, 2022-08-19:
@rdmark @akuker I suggest to close this ticket, because compilation times as such cannot be reduced. Just like in the past I agree that https://github.com/akuker/RASCSI/issues/740 would be the best solution, which makes this ticket essentially a duplicate.

rdmark, 2022-08-19:
Yes I agree that the one acceptable solution for this issue that I could imagine, would be to create and publish packaged binary releases.

Closing as Duplicate.

closed 2022-08-19 by rdmark