PiSCSI/piscsi #782: rascsi_version.cpp compile-time error

uweseimet, 2022-08-01:
With the current develop sources and gcc 11.3.0, at least on Linux x86_64:

rascsi_version.cpp:16:34: error: invalid digit "8" in octal constant
   16 | const int rascsi_minor_version = 08; // Month

I guess the latest release is also affected.

nucleogenic, 2022-08-01:
Same on `develop`, but `master` compiles OK.

Issue appears to be a result of develop version bump on commit https://github.com/akuker/RASCSI/commit/136e915f0c8eb58d9b8dddd15ef770de1624bd67.

Changing `08` to `8` appears to bypass the error, but I imagine this needs further attention to preserve the version number formatting?

akuker, 2022-08-02:
PR opened. Sorry about that!! Formatting will be preserved through the snprintf call later in this file. The extra 0 wasn't necessary.

https://github.com/akuker/RASCSI/pull/784

closed 2022-08-02 by uweseimet

uweseimet, 2022-08-02:
@akuker By the way, why is the latest release "22.07.02"? This contradicts the convention explained on https://github.com/akuker/RASCSI. Previous releases also did not all match the convention. "22.07" IMO would have to be the first July release according to the Gitvlow Workflow mentioned on https://github.com/akuker/RASCSI.
I recommend that the release versions either follow the convention defined for the project, or the conventions are changed. (To something which would then most likely be proprietary, which is usually not a good idea.)

rdmark, 2022-08-05:
@uweseimet I agree that there's a discrepancy between the README and what's the actual practice is in this project. Let me put up a PR for @akuker 's consideration.