PiSCSI/piscsi #790: Web interface not working

brianrudie2, 2022-08-15:
# Info

- Which version of Pi are you using: 2
- Which github revision of software:
- Which board version:amiga kit
- Which computer is the RaSCSI connected to: pi
- 

When I type in the rascsi s IP address to access the web interface I get the following message? and cannot see the web interface, I am a noob, any ideas?

Welcome to nginx!
If you see this page, the nginx web server is successfully installed and working. Further configuration is required.

For online documentation and support please refer to [nginx.org](http://nginx.org/).
Commercial support is available at [nginx.com](http://nginx.com/).

Thank you for using nginx.

rdmark, 2022-08-19:
@brianrudie2 This seems like an incomplete configuration of the Web Interface. It is more a troubleshooting situation than an issue in the software, so I would suggest you head over to the Discord server and ask for help in the support channel over there. https://discord.gg/PyS58u6

uweseimet, 2022-08-19:
@rdmark I would just like to point out something general: Anything discussed and resolved on Discord only will be lost for future users. Discord, Slack etc. are not useful to preserve information one can later easily look up ...

rdmark, 2022-08-19:
@brianrudie2 The first thing I would ask is: did you run the Web UI installation script in easyinstall.sh ?

rdmark, 2022-08-25:
Closing for now since the original reporter hasn't responded. Please feel free to reopen if the problem persists after re-running easyinstall.sh... You may also try one of the pre-built and configured images that we distribute with stable RaSCSI versions.

closed 2022-08-25 by rdmark