PiSCSI/piscsi #791: [Suggestion] - Make it easy to recover an image

TechGuyVincent, 2022-08-17:
Hi, 

Would love to have a simple way of recovering an image when things go wrong, or a dirty shutdown corrupt the drive (which happens frequently). Like a way to snapshot the image at each mount and be able to recover a previous snapshot instead of recovering the whole image. That would be awesome! 

Keep up the great work - love it!

rdmark, 2022-08-19:
@TechGuyVincent While not exactly what you're asking for, in the latest stable release there's a "Copy" function in the image management section of the Web UI. You could use that to manually make a copy every time before you mount it.

In the long term, there are plans to improve RaSCSI to make data corruption less likely. F.e. https://github.com/akuker/RASCSI/issues/335 as well as the addition of a supercap UPS hat https://github.com/akuker/RASCSI/wiki/supercapacitor-ups

uweseimet, 2022-08-19:
I suggest to close this ticket as a duplicate of https://github.com/akuker/RASCSI/issues/335. It's essentially the same problem/request, and we are already suffering from an increasing number of open tickets where something was started and then not followed up/abandoned.

closed 2022-08-19 by rdmark

lukecyca, 2022-09-18:
What about putting the underlying image storage on ZFS or BTRFS and using copy-on-write to make light-weight copies for snapshotting purposes?

This way you could quickly snapshot and branch images prior to making changes to a system. For example, prior to doing a risky/unsupported software upgrade on a classic mac. Currently we just have to make a copy of the entire image.